magnusja/libaums #2: Can not write random on file...

Shuhratjava, 2015-02-23:
Hi, 
I'm trying to write file by random accessing on it. There is RandomAccessFile class on java 7. I tried to write to sdcard directly by using that class. How can I do the same thing on this lib?
Best regards, Shukhrat

mjdev, 2015-02-24:
Hello,

you can randomly access any file via the read and write methods. You can spcify the offset (in bytes) where to begin with reading or writing.

Can you maybe give me a small example which does not work for you?

Best regards

Shuhratjava, 2015-02-24:
I have two methods with one of them I use for writing data, another one for seek for some position.
   @Override
            public void onWrite(byte[] buffer, int size) {
                Log.d("buffer size:", size + "");
                if (size < 0) {
                    try {
                        if (usbFile != null) {  
                            usbFile.close();
                        }
                        return;
                    } catch (IOException xc) {
                        xc.printStackTrace();
                    }
                }
                try {

                    if (usbFile != null) { 
                        usbFile.write(lastOffset, ByteBuffer.wrap(buffer));
                    }
                } catch (IOException e) {
                    e.printStackTrace();
                }
                lastOffset = lastOffset + size;

            }

            @Override
            public void onSeek(long offset, int whence) {
               
                        lastOffset = offset; 
            } 

At the end when I open the usb drive there is 0 bytes

mjdev, 2015-02-24:
Hi,

yes the buffer size affects the speed. The offset should be a multiple of the block size, which is usually 512 byte on usb mass storage. If it is not the data before the offset has to be read and then written back to the disk with the new data appended (which obvisouly takes longer). That is a problem which could be solved by implementing some caching.

I also explained that in my thesis if you would like to get more information on that.

What I would suggest is, use 4096 byte as buffer size.

Kind regards

Shuhratjava, 2015-02-26:
Why time of reading data before the offset takes such a long time? I see that for writing a buffer it takes 2 ms, at the time when for reading it takes around 880 ms?

closed 2015-03-02 by Shuhrat-java