magnusja/libaums #3: How to speed up writing time?

Shuhrat-java, 2015-03-02:
How can I increase speed of writing file on usb, on which part of code should I look for optimisation?

mjdev, 2015-03-04:
Hey,

first file you can look at is https://github.com/mjdev/libaums/blob/master/libaums/src/com/github/mjdev/libaums/driver/scsi/ScsiBlockDevice.java

Have a look at the methods read and write.

The optimal solution would be to implement an additional cache layer, which for example reads ahead or does not write immediately, but waits and then writes bigger chunks.

In the filesystem package (FAT32) there is also room for optimization.

Shuhrat-java, 2015-03-05:
Hey,
Thank you very much. I used caching logic, now I fill buffer with size of 1 MB and then write it to file. Speed dramatically increased ! 
Good luck

closed 2015-03-05 by Shuhrat-java

Shuhrat-java, 2015-04-11:
How can I get file creation time?

mjdev, 2015-04-11:
Currently there is no API für that. Nur I can add that next week. You can also try it via reflections or add the methods yourself and file a pull request.

mjdev, 2015-04-14:
Please have a look at my latest commit.

Shuhrat-java, 2015-04-15:
Thank you very much !!!