magnusja/libaums #5: Delete directory bug

nicolausYes, 2015-03-06:
This bug can be reproduced on delete operation very often. Or any other operation that initialize ClusterChain. It fails with OutOfMemoryError exception.

Usually, it is reproducible on deleting (or listing) directory with subdirectory inside that has some files.
For example.

* directory
  * subdirectory
      * file1
      * file2

If the directory without subdirectories is deleting in most of situations everything is ok.

#### What is in the backtrace.

```
 Process: xxx, PID: 14420
    java.lang.RuntimeException: An error occured while executing doInBackground()
            at android.os.AsyncTask$3.done(AsyncTask.java:300)
            at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
            at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
            at java.util.concurrent.FutureTask.run(FutureTask.java:242)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
            at java.lang.Thread.run(Thread.java:841)
     Caused by: java.lang.OutOfMemoryError
            at java.util.ArrayList.add(ArrayList.java:118)
            at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:121)
            at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:63)
            at com.github.mjdev.libaums.fs.fat32.FatDirectory.init(FatDirectory.java:154)
            at com.github.mjdev.libaums.fs.fat32.FatDirectory.listFiles(FatDirectory.java:442)
```

#### What is in the log before failing.
It is always the same. You can see that two blocks are repeating infinitely until there is no more free memory on device and then application is crashing.
The code is failing in this place in FAT.java.
```
result.add(currentCluster);
```

```
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 0
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 3
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/dalvikvm﹕ GC_FOR_ALLOC freed 512K, 12% free 17413K/19760K, paused 9ms, total 9ms
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 11
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=54752, transferBytes=16384, blockSize=512, transferBlocks=32, getdCbwDataTransferLength()=16384]
D/ScsiBlockDevice﹕ read time: 2
D/ScsiBlockDevice﹕ reading: ScsiRead10 [blockAddress=25050, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
D/ScsiBlockDevice﹕ read time: 1
```

mjdev, 2015-03-08:
Thank you very much (also for your pull request), I will have a look at this as soon as possible.

mjdev, 2015-03-13:
I could reproduce your problem with an empty file which had start cluster 0.

Can you give me some more information on what file this problem occurs or maybe even upload an image file?

nicolausYes, 2015-03-14:
In my case, it doesn't depend on types of files (I've just tried jpeg and mp3), their size, naming etc. And it can be stably reproduced in any cases.  I've tried three USB devices and it is reproduced on all of them.

Here are two examples of files structure, just extract them in the root folder of the USB and try to remove. If I removing subfolder (like 2015_03_14_16_05_57), it's ok, it's removing succesfully. The problem is with deleting root (AB8D69FA-8873-42D3-BB0E-4D53DA9B7A3F_storage) directory.

[example 1](https://dl.dropboxusercontent.com/u/20468014/AB8D69FA-8873-42D3-BB0E-4D53DA9B7A3F_storage_1.7z)
[example 2](https://dl.dropboxusercontent.com/u/20468014/AB8D69FA-8873-42D3-BB0E-4D53DA9B7A3F_storage.7z)

Thanks for help.

mjdev, 2015-03-26:
Hi Nicolaus,

sorry for my late response.

I tried to reproduce the problem with your data several times, but had no luck. What OS are you using to copy the data onto the USB drive? Can you give me any other additional information?

I also updated the master branch, can you give it a try and tell me if maybe the problem has been solved with the newest version?

nicolausYes, 2015-04-01:
I'm using Nexus 5 with android 4.4.4.
Usually there can be some small files in root folder near that_folder_with_long_name, but it seems doesn't matter.
I can provide any information you need :) This is how Memory Manager looks like while delete.
[image]

I've tried the code from the master, but the crash is still occurring.

Thanks for help.

mjdev, 2015-04-02:
Hello,

can you maybe prvide an image file for which the problem occurs? How to do that: https://wiki.archlinux.org/index.php/Disk_cloning under "Create disk image".

I can then put the image onto my flash drive with the exact file system (FAT and so on) like on your drive.

Kind regards.

mjdev, 2016-02-18:
Closed as this seems to be the same bug as in issue #9

closed 2016-02-18 by mjdev