magnusja/libaums #6: Partition size related problems on some devices .....

Shuhrat-java, 2015-04-01:
Hello, 
I have problems with some usb drives. During setting up usb device, drive partitions are being returned like  zero (0). What is the problem of that. I understand if it is not right file system. However I test with fat32 file system formatted usb drives.  I call init() method before getting partitions as provided in example code: 
   **
     * Sets the device up and shows the contents of the root directory.
     */
    private void setupDevice() {
        try { 
            device.init();
            if (device.getPartitions().size() != 0) {
//do your stuff
}

What is the cause of problem where partition size is 0?

mjdev, 2015-04-02:
Hello,

interesting problem, unfortunately I cannot say what is wrong. 

Can you maybe prvide an image file for which the problem occurs? How to do that: https://wiki.archlinux.org/index.php/Disk_cloning under "Create disk image".

Maybe you can also debug a bit on yourself and step into the init() method and look what is going on?

Kind regards.

mjdev, 2016-02-18:
Do you have any updates on this issue, or may I close this?

Thanks

Shuhrat-java, 2016-02-18:
I don't have that problem right now. Yes, maybe you can close this issue.

mjdev, 2016-02-18:
Thanks, if the problem occurs again, do not hesitate to open the issue again or create a new one.

closed 2016-02-18 by mjdev

Shuhrat-java, 2016-02-18:
Thank you also for your effort