magnusja/libaums #7: Expand support to include FAT12

YankTagpro, 2015-08-14:
What kind of effort would it take to expand support to include the FAT12 file system?

mjdev, 2015-08-16:
Hey,

just have a look here: https://github.com/waldheinz/fat32-lib/tree/master/src/main/java/de/waldheinz/fs 

That library has FAT12 and FAT16 support. I think it would also possible to use that library instead of mine pretty easy.

At the moment I do not have time to add FAT12 support sorry.

But I am curious, why do you need that? I thought that only FAT32 would be needed, so I just left FAT12 and FAT16 out :D

YankTagpro, 2015-08-20:
I needed it because the device I was integrating with was FAT12. My requirements were such that I only needed read-only capabilities for one specific file on the device, so it only took a few days. The biggest differences were the lack of partition tables and of course, the 12 bit FAT entries. 

Thank you for the library. It was very helpful!

closed 2015-08-20 by YankTagpro

Y-E-P, 2016-07-22:
@mjdev Hi,is there any library for android otg for support a FAT12? I wery need it. Or can you show example for use https://github.com/waldheinz/fat32-lib/tree/master/src/main/java/de/waldheinz/fs in android project?