magnusja/libaums #8: Support other file systems

Shuhrat-java, 2015-11-13:
1. How to know what kind of file system plugged usb drive has?
2. How to add support for NTFS?

mjdev, 2015-11-13:
Hey,

thanks for your question.
1. You can see that in the MBR in the partition table: https://en.wikipedia.org/wiki/NTFS For NTFS the identifier would be 0x07.
2. Basically someone has to code that functionality, either by using existing code, from for instance the Linux kernel or maybe the JNode OS (https://github.com/jnode/jnode/tree/master/fs/src/fs/org/jnode/fs/ntfs), or implementing the whole thing himself.

Kind regards

mjdev, 2016-02-18:
So far I do not have any plan to implement other fs types, but any contribution is highly appreciated.

closed 2016-02-18 by mjdev