magnusja/libaums #9: Delete all files bug.

graynote, 2015-11-26:
When I try to delete all files in root directory, it fails to delete a last file. 
It looks like there are 2 problems in the code.

1. FatDirectory.init() : 
    if (entries.size() == 0)
	readEntries(); -> this code read a last entry again.. 

2. FatDirectory.write()
If entries.size() == 0 and writeVolumeLabel == false, it does not write well.

mjdev, 2015-11-26:
Hey,

thanks for your feedback.

If you have already figured out which code lines might cause the trouble, can you maybe fix it and just open a pull request? I am really busy at the moment.

Thanks

Magnus

graynote, 2015-11-27:
My workaround is not a good solution. I created VolumeLabel to write something when there is no entry. I have no idea how to fix it perfectly yet.

mjdev, 2015-12-15:
Hey,

can you test if the commit in branch bugfix/issue-9 solves your problem?

Kind regards
Magnus

tliebeck, 2016-02-18:
WARNING/DISCLAIMER: I have essentially no knowledgeable of the FAT filesystem.  The following is based only on reviewing the code, please be extremely careful taking any advice from me on this subject.

I believe this problem may be caused by FatDirectory.init().  The method contains the following code:

		if (entries.size() == 0)
			readEntries();

The FatDirectory.write() method calls init().  In the case that FatDirectory.write() is being called and no files are present, entries.size() will be 0.  Thus readEntries() call will thus put the entries object into an invalid state before writing.

I hacked my own version to avoid having init() call readEntries() in this specific scenario, and it fixes the problem.

mjdev, 2016-02-18:
Hey tlieback thanks for your hint, can you maybe post how you solved the problem? Help is really appreciated. Thanks!

mjdev, 2016-02-18:
@tliebeck In general I find it strange that reading the entries again will corrupt the directory in memory representation, because it just reads zero entries, right?

tliebeck, 2016-02-18:
I was under the impression it was re-reading the old state, i.e., that it would read it back as "one entry" and then rewrite that.

tliebeck, 2016-02-18:
Regarding the code changes, I modified FatDirectory.init() as follows:

```
	private void init(boolean readEntriesIfRequired) throws IOException {
		if (chain == null) {
			chain = new ClusterChain(entry.getStartCluster(), blockDevice, fat, bootSector);
		}

		if (readEntriesIfRequired && entries.size() == 0)
			readEntries();
	}

```

The "readEntriesIfRequired" parameter is added.  The write() method calls init(false).  All other methods call init(true).

This is probably not the best solution, i.e. I don't know if there are other cases where readEntriesIfRequired should be false.  

Perhaps there should be a instance variable boolean flag that indicates whether the "entries" collection is up-to-date, given that it being empty is a valid state and re-loading it could place it in an incorrect state?

mjdev, 2016-02-18:
@tliebeck You are completely right, deleting all entries and then reading them again is complete nonsense, although the misbehaviour was really strange. For me the app crashed with some OutOfMemoryExceptions.

Can you have a look at current develop branch and tell me if the bug is fixed now? it works for me.

tliebeck, 2016-02-19:
I'm seeing major problems with the updated code, creating new directories now results in filesystem corruption.  My hacked version works fine here, but the current "develop" is a corrupting immediately after directory creation.

Result of making a directory off the root called "alpha" and opening it (my app):
[image]

mjdev, 2016-03-20:
@tliebeck Thanks for the help, I think I fixed this bug (also mentioned in issue #18), in current develop, please review and tell me if it works for you, tanks.

tliebeck, 2016-05-19:
As of today's checkout (May 19 2016), I still see this bug as present.  It's correct if I use  my modified init() method:

	/**
	 * Initializes the {@link FatDirectory}. Creates the cluster chain if needed
	 * and reads all entries from the cluster chain.
	 * 
	 * @throws IOException
	 *             If reading from the device fails.
	 */
	private void init(boolean readEntriesIfRequired) throws IOException {
		if (chain == null) {
			chain = new ClusterChain(entry.getStartCluster(), blockDevice, fat, bootSector);
		}

		// entries is allocated here
		// an exception will be thrown if entries is used before the directory has been initialised
		// use of uninitialised entries can lead to data loss!
		if(entries == null) {
			entries = new ArrayList<FatLfnDirectoryEntry>();
		}

		// only read entries if we have no entries
		// otherwise newly created directories (. and ..) will read trash data
		if(readEntriesIfRequired && entries.size() == 0) {
			readEntries();
		}
	}

All calls to init() are passed true except for the one call made from the write() method.  

In the write() method, when deleting the last file, entries.size() will return 0 (all files are deleted). 
In the current version, this will trigger readEntries() to be invoked, which makes things out of of sync, corrupting the filesystem.

Full copy of modified FatDirectory class attached.

[FatDirectory.java.zip](https://github.com/mjdev/libaums/files/272038/FatDirectory.java.zip)

mjdev, 2016-05-20:
Hey @tliebeck thank you for the feedback. Pretty embarrassing, that I made this mistake again :D
I will have a look at it. Maybe I find a more elegant solution then passing this parameter all the time ;)

mjdev, 2016-06-19:
I hope this is fixed now

closed 2016-06-19 by mjdev