magnusja/libaums #10: Issue to Access USB Mass Storage over Type C

anuragcm, 2016-01-07:
Hello,

I am developing an application to access USB mass storage connected to Android phone. I am using libaums library. When I run application in devices with micro USB interface, I am able to access and explore the drive. When I run application in device with USB Type C connector (Nexus 5X and Nexus 6P), I face the issue. I am seeing the follwing exception in the logcat. 

01-06 12:03:29.181 30348 30348 E TYPEC     : error setting up device
01-06 12:03:29.181 30348 30348 E TYPEC     : java.io.IOException: reading failed!
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:144)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.libaums.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.java:77)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.java:312)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.java:268)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.kt.TYPEC.setupDevice(TYPEC.java:409)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.kt.TYPEC.access$600(TYPEC.java:92)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.test.kt.TYPEC$7.onReceive(TYPEC.java:315)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:881)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at android.os.Handler.handleCallback(Handler.java:739)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at android.os.Handler.dispatchMessage(Handler.java:95)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at android.os.Looper.loop(Looper.java:148)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at android.app.ActivityThread.main(ActivityThread.java:5417)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at java.lang.reflect.Method.invoke(Native Method)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
01-06 12:03:29.181 30348 30348 E TYPEC     : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

It looks like device doesn't respond to SCSI inquiry.

Any help to resolve the issue is highly appreciated.

Thanks,
Anu

mjdev, 2016-01-08:
Hey Anu,

thanks for your feedback. 

Can you tell me what OTG cable you are using? I could not find any OTG cable for Type C. 
Unfortunately this is really hard to test for me because I do not have USB 3.x hardware. Can you maybe provide some additional log output?
Can you test it with Nexus Media Importer (If you have it) or with Total Commander and a plugin https://play.google.com/store/apps/details?id=de.hechler.tcplugins.usbstick&hl=de or https://play.google.com/store/apps/details?id=com.paragon.tcplugins_ntfs_ro

Kind regards,
Magnus

BTW, there is no need to mail me extra, I get a notification email when issues are created ;)

KevinXiCN, 2016-01-18:
Hello Magnus,

It's Kevin, I am working for Picture Keeper with Shawney Moore.

If you recall, we were using your libaums in our Picture Keeper Android app: https://play.google.com/store/apps/details?id=com.simplifieditproducts.picturekeeperconnect

It works great, until we recently test our app with a USB 3.0 drive. I think the issue is similar to this one. We were able to mount the drive when we connect with a USB 2.0 disk via OTG cable. But we can't do the same when we connect with a USB 3.0 disk. PS: I am testing with Galaxy S5.

Here are the results for the 3 apps I tested. All of them are trying to unmount the disk from Android system and mount the disk with their own code.

- Picture Keeper: blocked when connect / mount the drive.
- Nexus Media Importer: blocked when connect / mount the drive.
- Total Commander (with Paragon): worked properly.

Do you think it is a hardware issue, or it can be handled within libaums library?

Thanks,
Kevin

[image]
[image]

KevinXiCN, 2016-01-18:
I am able to debug this issue via WiFi so that I can keep USB 3.0 disk connecting with the Galaxy S5. Just want to share some more details to you. 

As you can see from the screenshot below, the 'Inquiry' command was sent to drive successfully (31 bytes), then the library was waiting for SCSI response, its current excepted response length was 36 bytes but actually only 19 bytes were read. The app was blocked to wait for more incoming data.

[image]

mjdev, 2016-01-18:
Hey guys,

thanks for the feedback.

Unfortunately this seems to be a bigger fix: https://en.wikipedia.org/wiki/USB_Attached_SCSI

I do not have an USB 3 hardware, so I cannot test it, yet. I also have only a limited amount of time, so if someone of you could do the fix and then file a pull request, that would the highly appreciated.

mjdev, 2016-02-09:
Merged into develop, should be fixed now.

closed 2016-02-09 by mjdev