magnusja/libaums #11: Capacity of USB drive.

Shuhrat-java, 2016-01-26:
Hello.
How can I know the capacity of USB drive? Seems there is not any API which gives that feature. I need to know how much free space left for writing

mjdev, 2016-01-26:
Hey there,

you are right, this is currently not exposed by the API,but supported in some way in FAT32. I will take care of that. In the meantime you can easily add that feature yourself by delegating the required method calls down to the FAT32 layer or by nasty casting of the FSFileSystem things.

Shuhrat-java, 2016-01-27:
Thanks. However I have no idea how to do that

mjdev, 2016-02-09:
Hey, please have a look at current develop, introduced new methods in interface FileSystem.

closed 2016-02-09 by mjdev

Shuhrat-java, 2016-02-10:
I don't see any new method in FileSystem interface?