magnusja/libaums #13: Writing on USB drive failing sometimes...

Shuhrat-java, 2016-02-10:
I'm using this lib for writing  and reading video files. Sometimes when I write by part I get exceptions shown in below stack:
```
02-10 15:34:06.980    2802-6916/? W/System.err﹕ java.io.IOException: reading failed!
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:149)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:215)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.partition.Partition.read(Partition.java:133)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.fs.fat32.FAT.alloc(FAT.java:202)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.fs.fat32.ClusterChain.setClusters(ClusterChain.java:199)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.fs.fat32.ClusterChain.setLength(ClusterChain.java:230)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.fs.fat32.FatFile.setLength(FatFile.java:151)
02-10 15:34:06.980    2802-6916/? W/System.err﹕ at com.github.mjdev.libaums.fs.fat32.FatFile.write(FatFile.java:167)
```
I don't have any idea why this happens, because sometimes it works OK.

mjdev, 2016-02-10:
Hey,

can you maybe give some further information on which hardware your are using?
In general this issue means that the USB device did not respond with an correct answer to a command, which is really strange and hard to analyze.

Shuhrat-java, 2016-02-10:
I'm using Sandisk Ultra USB 3.0 16GB and I tested with Nexus 7 and Lenovo Yoga Tablet 8". In both tablets I have problem. On lenovo tablet often than on Nexus.

mjdev, 2016-02-18:
Can you maybe test a different drive? Preferably USB 2.0? 

Thanks

Shuhrat-java, 2016-02-18:
I use different devices and all of them have that issue.

diogowbrito, 2016-06-09:
I'm also getting this issue with a Sandisk Dual Drive 16GB, doing regular copy operations.

I also see multiple times on logcat the `ECONNRESET` on `NanoHTTPD`

Stack Traces:
```
E/ScsiBlockDevice: Writing all bytes on command ScsiRead10 [blockAddress=411706, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024] failed!
E/UsbFile: IOException
           java.io.IOException: reading failed!
               at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:149)
               at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:215)
               at com.github.mjdev.libaums.partition.Partition.read(Partition.java:133)
               at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:134)
               at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:63)
               at com.github.mjdev.libaums.fs.fat32.FatDirectory.init(FatDirectory.java:153)
               at com.github.mjdev.libaums.fs.fat32.FatDirectory.search(FatDirectory.java:420)
               at com.github.mjdev.libaums.fs.fat32.FatDirectory.search(FatDirectory.java:444)
               at com.github.mjdev.libaums.server.http.UsbFileHttpServer.serve(UsbFileHttpServer.java:102)
               at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:840)
               at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:189)
               at java.lang.Thread.run(Thread.java:818)

```

```
E/fi.iki.elonen.NanoHTTPD: Could not send response to the client
                           java.net.SocketException: sendto failed: ECONNRESET (Connection reset by peer)
                               at libcore.io.IoBridge.maybeThrowAfterSendto(IoBridge.java:542)
                               at libcore.io.IoBridge.sendto(IoBridge.java:511)
                               at java.net.PlainSocketImpl.write(PlainSocketImpl.java:500)
                               at java.net.PlainSocketImpl.access$100(PlainSocketImpl.java:37)
                               at java.net.PlainSocketImpl$PlainSocketOutputStream.write(PlainSocketImpl.java:266)
                               at java.io.OutputStream.write(OutputStream.java:82)
                               at fi.iki.elonen.NanoHTTPD$Response$ChunkedOutputStream.write(NanoHTTPD.java:1259)
                               at fi.iki.elonen.NanoHTTPD$Response$ChunkedOutputStream.write(NanoHTTPD.java:1247)
                               at java.util.zip.GZIPOutputStream.<init>(GZIPOutputStream.java:84)
                               at java.util.zip.GZIPOutputStream.<init>(GZIPOutputStream.java:54)
                               at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectEncoding(NanoHTTPD.java:1449)
                               at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectTransferAndEncoding(NanoHTTPD.java:1440)
                               at fi.iki.elonen.NanoHTTPD$Response.send(NanoHTTPD.java:1429)
                               at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:852)
                               at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:189)
                               at java.lang.Thread.run(Thread.java:818)
                            Caused by: android.system.ErrnoException: sendto failed: ECONNRESET (Connection reset by peer)
                               at libcore.io.Posix.sendtoBytes(Native Method)
                               at libcore.io.Posix.sendto(Posix.java:211)
                               at libcore.io.BlockGuardOs.sendto(BlockGuardOs.java:278)
                               at libcore.io.IoBridge.sendto(IoBridge.java:509)
                               at java.net.PlainSocketImpl.write(PlainSocketImpl.java:500) 
                               at java.net.PlainSocketImpl.access$100(PlainSocketImpl.java:37) 
                               at java.net.PlainSocketImpl$PlainSocketOutputStream.write(PlainSocketImpl.java:266) 
                               at java.io.OutputStream.write(OutputStream.java:82) 
                               at fi.iki.elonen.NanoHTTPD$Response$ChunkedOutputStream.write(NanoHTTPD.java:1259) 
                               at fi.iki.elonen.NanoHTTPD$Response$ChunkedOutputStream.write(NanoHTTPD.java:1247) 
                               at java.util.zip.GZIPOutputStream.<init>(GZIPOutputStream.java:84) 
                               at java.util.zip.GZIPOutputStream.<init>(GZIPOutputStream.java:54) 
                               at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectEncoding(NanoHTTPD.java:1449) 
                               at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectTransferAndEncoding(NanoHTTPD.java:1440) 
                               at fi.iki.elonen.NanoHTTPD$Response.send(NanoHTTPD.java:1429) 
                               at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:852) 
                               at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:189) 
                               at java.lang.Thread.run(Thread.java:818) 
```

mjdev, 2016-06-19:
@diogowbrito The latter one should not be a problem. Depending on the browser this happens, especially when streaming content, like videos.

46472761, 2017-04-06:
JAMES LACONIA DAVIS,TRIPLE ELITE,46472761

closed 2017-07-03 by magnusja