magnusja/libaums #14: FS Corruption when using UsbFile.moveTo()

tliebeck, 2016-02-18:
I'm seeing filesystem corruption when moving directories using UsbFile.moveTo().

I've been testing this by creating the following directory structure (in the root directory of the filesystem):

```
Alpha <DIR>
   File1.txt
Beta <DIR>
    File2.txt
```

If I invoke alphaUsbFile.moveTo(betaUsbFile), the directory structure will appear as follows:

```
Beta <DIR>
   Alpha <DIR>
      File1.txt
```

That is, File2.txt is inexplicably deleted.

Further file operations will, over time, result in total destruction of the filesystem, i.e. file listings return hierarchies of garbage directories/files.  For example. deleting the "Beta" directory tends to do so.

Have seen this behavior with both "master" and "develop" branches.  Thanks for any suggestions, and also generally for your excellent work on this library!

mjdev, 2016-02-18:
Hey @tliebeck, thanks for your feedback. Do you think this issue could be related to issue #9?

I will have a look at it as soon as I can.

mjdev, 2016-02-18:
Hey @tliebeck I quickly tried that with the example app and the most recent develop i.e. issue #9 fixed. So far this procedure works for me. Can you maybe post an image from the drives initial state which then leads to this bug?

Thanks

tliebeck, 2016-02-19:
I've just solved this by adding calls to init() in FatDirectory.move/moveTo():

```
		parent.init();
		destinationDir.init();

		// now the actual magic happens!
		parent.removeEntry(entry);
		destinationDir.addEntry(entry, entry.getActualEntry());
```

Looks like one or both is not initialized, resulting in the lost information.  My read of your code is that init() should be safe to call at any time, correct...as it's supposed to do nothing if it's already initialized?

As far as the initial state of the drive, just go with the two folders each with one file in them configuration shown in the first post.  For me it reliably fails every time if I move "alpha" into "beta" (fixed now with the above change to the move/moveTo() methods).

mjdev, 2016-03-20:
Thanks for your help, I applied your changes to develop.

closed 2016-03-20 by mjdev