magnusja/libaums #15: Video file corrupting when writing in real time...

Shuhrat-java, 2016-03-04:
Hello,
In my project I'm writing real time encoded mp4 video file to USB drive. At the end of recording mp4 file should have some header information. For that ffmpeg seeks and writes some bytes. Some of them are 4 some of them 40, 64, 233 etc. However in some cases file is being corrupted. I tried to write with same logic to android's sdcard and it is going everytime with success but with USB sometimes fails. Is it related to cluster size and something else which is the cause? Could you please explain it. I don't know how FAT should behaviour when writing file. Thanks in advance

mjdev, 2016-03-07:
Hey,
thanks for reporting, this is definitely a bug, I think seeking to the front and writing there is just not really implemented :/

Shuhrat-java, 2016-03-08:
However seeking goes backward and forward not only to the front. After seeking bytes will be overwritten on backward

mjdev, 2016-03-21:
Hey @Shuhrat-java can you please share some more code, in my previous posting I made a mistake, seeking should work fine.
Maybe you can paste the interaction of ffmpeg and libaums.

mjdev, 2016-05-08:
Closed due to inactivity, if you encounter the error again, please reopen.

closed 2016-05-08 by mjdev