magnusja/libaums #18: createDirectory function create that can not be deleted file.

vlog-yamamoto, 2016-03-17:
createDirectory function creates a cannot delete file under the directory it created.
There is also a problem in create directory's volume label from log.

Operation is as follows:
1. createDirectory "a" in Android
2. USB device detach from Android
3. USB device attach to PC (My PC is mac)
4. Delete Directory "a" in PC
5. USB device detach from PC use eject function.
6. USB device attach to Android 
7. createDirectory "a" create that can not be deleted file.

log is following in libaums-develop application.
```
03-17 10:39:25.272  18396-18396/com.github.mjdev.usbfileman I/art﹕ Late-enabling -Xcheck:jni
03-17 10:39:25.392  18396-18396/com.github.mjdev.usbfileman W/MainActivity﹕ no device found!
03-17 10:39:25.401  18396-18416/com.github.mjdev.usbfileman D/OpenGLRenderer﹕ Use EGL_SWAP_BEHAVIOR_PRESERVED: true
03-17 10:39:25.406  18396-18396/com.github.mjdev.usbfileman D/Atlas﹕ Validating map...
03-17 10:39:25.441  18396-18416/com.github.mjdev.usbfileman I/Adreno-EGL﹕ <qeglDrvAPI_eglInitialize:379>: QUALCOMM Build: 01/14/15, ab0075f, Id3510ff6dc
03-17 10:39:25.443  18396-18416/com.github.mjdev.usbfileman I/OpenGLRenderer﹕ Initialized EGL, version 1.4
03-17 10:39:25.476  18396-18416/com.github.mjdev.usbfileman D/OpenGLRenderer﹕ Enabling debug mode 0
03-17 10:39:25.505  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ on service connected ComponentInfo{com.github.mjdev.usbfileman/com.github.mjdev.libaums.server.http.UsbFileHttpServerService}
03-17 10:39:32.024  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ USB device attached
03-17 10:39:32.063  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb device: UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=1921,mProductId=53232,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=SanDisk,mProductName=iXpand Flash Drive,mSerialNumber=AB0123456789,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=250,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
03-17 10:39:32.063  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
03-17 10:39:32.063  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb endpoint: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
03-17 10:39:32.063  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb endpoint: UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]
03-17 10:39:33.919  18396-18396/com.github.mjdev.usbfileman D/UsbMassStorageDevice﹕ setup device
03-17 10:39:33.928  18396-18396/com.github.mjdev.usbfileman D/ScsiBlockDevice﹕ inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=0, responseDataFormat=1]
03-17 10:39:33.934  18396-18396/com.github.mjdev.usbfileman I/ScsiBlockDevice﹕ Block size: 512
03-17 10:39:33.934  18396-18396/com.github.mjdev.usbfileman I/ScsiBlockDevice﹕ Last block address: 62333951
03-17 10:39:33.943  18396-18396/com.github.mjdev.usbfileman I/FAT﹕ fat is mirrored, fat count: 2
03-17 10:39:33.961  18396-18396/com.github.mjdev.usbfileman D/FatDirectory﹕ volume label: USBSTORAGE
03-17 10:39:33.961  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ Capacity: 31910789120
03-17 10:39:33.961  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ Occupied Space: 58884096
03-17 10:39:33.961  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ Free Space: 31851905024
03-17 10:39:37.218  18396-18396/com.github.mjdev.usbfileman W/art﹕ Before Android 4.1, method int android.support.v7.widget.ListViewCompat.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
03-17 10:39:47.829  18396-18396/com.github.mjdev.usbfileman D/FsInfoStructure﹕ writing to device
03-17 10:39:47.833  18396-18396/com.github.mjdev.usbfileman I/FAT﹕ allocating clusters finished
03-17 10:39:47.833  18396-18396/com.github.mjdev.usbfileman D/FatDirectory﹕ adding entry: [FatLfnDirectoryEntry getName()=a] with short name: A
03-17 10:39:47.834  18396-18396/com.github.mjdev.usbfileman W/ScsiBlockDevice﹕ we have to round up size to next block sector
03-17 10:39:47.846  18396-18396/com.github.mjdev.usbfileman W/ScsiBlockDevice﹕ we have to round up size to next block sector
03-17 10:39:58.429  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ USB device detached
03-17 10:39:58.429  18396-18396/com.github.mjdev.usbfileman D/UsbMassStorageDevice﹕ close device
03-17 10:39:58.429  18396-18396/com.github.mjdev.usbfileman E/UsbMassStorageDevice﹕ could not release interface!
03-17 10:39:58.429  18396-18396/com.github.mjdev.usbfileman D/UsbDeviceConnectionJNI﹕ close
03-17 10:39:58.439  18396-18396/com.github.mjdev.usbfileman W/MainActivity﹕ no device found!
03-17 10:40:38.407  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ USB device attached
03-17 10:40:38.438  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb device: UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=1921,mProductId=53232,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=SanDisk,mProductName=iXpand Flash Drive,mSerialNumber=AB0123456789,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=250,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
03-17 10:40:38.438  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
03-17 10:40:38.438  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb endpoint: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
03-17 10:40:38.438  18396-18396/com.github.mjdev.usbfileman I/UsbMassStorageDevice﹕ found usb endpoint: UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]
03-17 10:40:40.030  18396-18396/com.github.mjdev.usbfileman D/UsbMassStorageDevice﹕ setup device
03-17 10:40:40.035  18396-18396/com.github.mjdev.usbfileman D/ScsiBlockDevice﹕ inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=0, responseDataFormat=1]
03-17 10:40:40.037  18396-18396/com.github.mjdev.usbfileman I/ScsiBlockDevice﹕ Block size: 512
03-17 10:40:40.037  18396-18396/com.github.mjdev.usbfileman I/ScsiBlockDevice﹕ Last block address: 62333951
03-17 10:40:40.045  18396-18396/com.github.mjdev.usbfileman I/FAT﹕ fat is mirrored, fat count: 2
03-17 10:40:40.059  18396-18396/com.github.mjdev.usbfileman D/FatDirectory﹕ volume label: USBSTORAGE
03-17 10:40:40.060  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ Capacity: 31910789120
03-17 10:40:40.060  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ Occupied Space: 58949632
03-17 10:40:40.060  18396-18396/com.github.mjdev.usbfileman D/MainActivity﹕ Free Space: 31851839488
03-17 10:40:45.406  18396-18416/com.github.mjdev.usbfileman D/OpenGLRenderer﹕ endAllStagingAnimators on 0xb4bff200 (RippleDrawable) with handle 0xb4b73e40
03-17 10:40:51.984  18396-18396/com.github.mjdev.usbfileman D/FsInfoStructure﹕ writing to device
03-17 10:40:51.988  18396-18396/com.github.mjdev.usbfileman I/FAT﹕ allocating clusters finished
03-17 10:40:51.988  18396-18396/com.github.mjdev.usbfileman D/FatDirectory﹕ adding entry: [FatLfnDirectoryEntry getName()=a] with short name: A
03-17 10:40:52.001  18396-18396/com.github.mjdev.usbfileman W/FatDirectory﹕ volume label in non root dir!
03-17 10:40:52.001  18396-18396/com.github.mjdev.usbfileman D/FatDirectory﹕ volume label: ﾋ
03-17 10:40:52.001  18396-18396/com.github.mjdev.usbfileman W/ScsiBlockDevice﹕ we have to round up size to next block sector
```

PC's ls log
```
vLog-no-MacBook-Pro-2:~ yamamoto$ ls /Volumes/USBSTORAGE/a/
0bâ?␀␀␀ÿ.ÿ?␀			”Xœ‘Z¬oj.`Äà
```

Device
Nexus 5 Ver.5.1.1
Nexus 6 Ver.6.0.1 

UsbDrive
iXpand Flash Drive 32GB 
SanDisk Ultra USB 3.0 16GB.

vlog-yamamoto, 2016-03-17:
When listFiles function do after the createDirectory, sometimes listFiles is not 0.

```
UsbFile currentUsbFile = getRootDirectory();

UsbFile usbFile = currentUsbFile.createDirectory(name);

UsbFile[] usbFiles = nextUsbFile.listFiles();
if (usbFiles != null && usbFiles.length > 0) {
		Timber.e("createDirectory: usbfile[%d]", usbFiles.length);
}
```

mjdev, 2016-03-17:
Hey, 

thanks for reporting. Currently I am pretty busy, thus I cannot say when I can have a look at this.

vlog-yamamoto, 2016-03-18:
I tried the following in the Mac PC and Windows PC.

1. USB device attach to PC
2. USB device detach from PC use eject function.
3. USB device attach to Android
4. createDirectory function do.

When attach to Mac PC createDirectory function sometimes create that cannot be deleted. 
But When attach to Windows PC createDirectory function did not create that cannot be deleted. 

It might have occurred from usb storage mount to mac pc.

vlog-yamamoto, 2016-03-18:
It is seem that FatDirectory class createDirectory function read dirty memory.

createDirectory function result.entries.size() is two. result.entries is  "." and "..".
But sometime result.entries.size() not 2 after result.write().
write() is init() at first. 

```
    private void init() throws IOException {
        if (chain == null) {
            chain = new ClusterChain(entry.getStartCluster(), blockDevice, fat, bootSector);
            readEntries();
        }
    }
```

Is is createDirectory function's "result" is chain = null, So readEntries do.
I thinkk that When readEntries read dirty memory, volume label Log or addEntry is happend.

mjdev, 2016-03-20:
Should be fixed in current develop, please review and reopen this issue if not.

closed 2016-03-20 by mjdev

vlog-yamamoto, 2016-03-22:
createDirectory function only do in root directory. 

```
1.USB device attach to mac
2.USB device detach from mac use eject function.
3.USB device attach to Android
4.createDirectory function do in root directory (ex [root directory]/a)
5.USB device detach from Android
6.USB device attach to mac
7.ls command do.
```
The following problems are occurring often.

No.1 not find "." and ".." file.
```
$ ls -la /Volumes/USBSTORAGE/a/
```

No.2 create file.
```
$ ls -la /Volumes/USBSTORAGE/b/
ls: BF0541E4.-06: Input/output error
ls: E8DC␀␀␀␀.␀␀␀: No such file or directory
```

No.3 ".." time is error.
```
$ ls -la /Volumes/USBSTORAGE/h/ 
total 64
drwxrwxrwx  1 yamamoto  staff  16384  3 22 20:12 .
drwxrwxrwx@ 1 yamamoto  staff  16384  1  1  1970 ..
```

When createDirectory function do in not root directory , It seems not to occur
ex. [root directory]/a/b seems not to occur. 
    [root directory]/a is occur.

After createDirectory function do, It click createDirectory.
When show createDirectory, It seems not to occur No.1 and No2.

mjdev, 2016-03-22:
Can you reproduce this behaviour with the example app included in the repository?

vlog-yamamoto, 2016-03-23:
I use Usbfileman at developer brunch.
When only createDirectory do in root directory, problems are occurring often.

My application is copy file after createDirectory function. 
It seems not to occur after #18.

mjdev, 2016-03-23:
Okay I can reproduce No. 3 now.

For the others did you try a different Android device and did you ensure waiting enough time after detaching from the Android device (so data was actually written completely)?

vlog-yamamoto, 2016-03-24:
I was testing with Nexus5.
When I change device to Nexus6, problems are occurring.

When I change UsbMassStorage, I can reproduce only No.3.
The cause of No.1 and No.2 may be USB storage.

mjdev, 2016-03-24:
Okay I currently do not know why no. 3 is happening, but I will have a look asap.

Which of the flash drives is making problems? (iXpand Flash Drive 32GB or SanDisk Ultra USB 3.0 16GB?)

vlog-yamamoto, 2016-03-24:
iXpand Flash Drive 32GB is making problems.