magnusja/libaums #19: NullPointerException when calling root directorys getName() method

jmue, 2016-03-17:
When a UsbFile is a root directory it's `entry` class member is `null`. So calling method `getName()` will result in a NullPointerException.

`java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String com.github.mjdev.libaums.fs.fat32.FatLfnDirectoryEntry.getName()' on a null object reference`

mjdev, 2016-03-17:
Why would you do this on the root directory?

jmue, 2016-03-17:
same for method `lastModified()`

jmue, 2016-03-17:
I'm calling this on Interface `UsbFile`, my logic does not know if this specific element is the root directory.

mjdev, 2016-03-17:
I see, as a workaround until this is fixed you could check if the directory is the root directory via:
`if(usbFile.getParent() == null) 
    // is root`

mjdev, 2016-03-21:
Resolved in current develop, please use isRoot() to avoid querying root directory

closed 2016-03-21 by mjdev