magnusja/libaums #20: The sample file manager doesn't work with Android M

greddyatpt, 2016-03-24:
I downloaded the code yesterday and tried to run the file manager app with Android M as the target.  It crashed.  Haven't analyzed it yet but it could be something to do with the new runtime permissions.  Here's the log:

03-24 01:09:41.037 32060-32060/? W/MemoryReductionHack: Couldn't update the Loader
03-24 01:09:41.060 32060-32080/? I/GMPM: App measurement is starting up
03-24 01:09:41.082 32060-32080/? E/GMPM: getGoogleAppId failed with status: 10
03-24 01:09:41.083 32060-32080/? E/GMPM: Uploading is not possible. App measurement disabled
03-24 01:09:41.139 32060-32060/? W/StaticBindingVerifier: Verify
03-24 01:09:41.167 32060-32088/? W/LightSharedPreferencesImpl: Failed to load preference file from Disk!
                                                               java.io.FileNotFoundException: /data/user/0/com.facebook.katana/app_light_prefs/com.facebook.katana/analytics_flexible_sampling_policy: open failed: ENOENT (No such file or directory)
                                                                   at libcore.io.IoBridge.open(IoBridge.java:452)
                                                                   at java.io.FileInputStream.<init>(FileInputStream.java:76)
                                                                   at com.facebook.crudolib.prefs.LightSharedPreferencesStorage.a(ufi_enabled/:56)
                                                                   at com.facebook.crudolib.prefs.LightSharedPreferencesImpl$1.run(ui thread:61)
                                                                   at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:423)
                                                                   at com.facebook.common.executors.WrappingExecutorService$1.run(verified_server_address:77)
                                                                   at com.facebook.common.executors.DefaultConstrainedListeningExecutorService$Worker.run(vibrator:327)
                                                                   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                   at com.facebook.common.executors.NamedThreadFactory$1.run(video-player-readable-cache:42)
                                                                   at java.lang.Thread.run(Thread.java:818)
                                                                Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
                                                                   at libcore.io.Posix.open(Native Method)
                                                                   at libcore.io.BlockGuardOs.open(BlockGuardOs.java:186)
                                                                   at libcore.io.IoBridge.open(IoBridge.java:438)
                                                                   at java.io.FileInputStream.<init>(FileInputStream.java:76) 
                                                                   at com.facebook.crudolib.prefs.LightSharedPreferencesStorage.a(ufi_enabled/:56) 
                                                                   at com.facebook.crudolib.prefs.LightSharedPreferencesImpl$1.run(ui thread:61) 
                                                                   at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:423) 
                                                                   at com.facebook.common.executors.WrappingExecutorService$1.run(verified_server_address:77) 
                                                                   at com.facebook.common.executors.DefaultConstrainedListeningExecutorService$Worker.run(vibrator:327) 
                                                                   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
                                                                   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
                                                                   at com.facebook.common.executors.NamedThreadFactory$1.run(video-player-readable-cache:42) 
                                                                   at java.lang.Thread.run(Thread.java:818)

mjdev, 2016-03-24:
Hey, thanks for reporting.

Just go into settings and allow USB access for the example app.

greddyatpt, 2016-03-24:
Thanks.  I did that but that didn't make a difference.  (I'd like to tell you about my project at SanDisk.  Where do I send the email?)

03-24 08:40:27.240 5650-5650/? I/art: Late-enabling -Xcheck:jni
03-24 08:40:27.316 5650-5650/com.github.mjdev.usbfileman W/System: ClassLoader referenced unknown path: /data/app/com.github.mjdev.usbfileman-2/lib/arm64
03-24 08:40:27.377 5650-5650/com.github.mjdev.usbfileman I/UsbMassStorageDevice: found usb device: UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=3034,mProductId=790,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=Generic,mProductName=USB3.0-CRW,mVersion=2.16,mSerialNumber=20120501030900000,mConfigurations=[
                                                                                 UsbConfiguration[mId=1,mName=CARD READER,mAttributes=128,mMaxPower=250,mInterfaces=[
                                                                                 UsbInterface[mId=0,mAlternateSetting=0,mName=Bulk-In, Bulk-Out, Interface,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
                                                                                 UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
                                                                                 UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
03-24 08:40:27.377 5650-5650/com.github.mjdev.usbfileman I/UsbMassStorageDevice: found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=Bulk-In, Bulk-Out, Interface,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
                                                                                 UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
                                                                                 UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
03-24 08:40:27.377 5650-5650/com.github.mjdev.usbfileman I/UsbMassStorageDevice: found usb endpoint: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
03-24 08:40:27.377 5650-5650/com.github.mjdev.usbfileman I/UsbMassStorageDevice: found usb endpoint: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
03-24 08:40:27.387 5650-5673/com.github.mjdev.usbfileman D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
03-24 08:40:27.395 5650-5650/com.github.mjdev.usbfileman D/UsbMassStorageDevice: setup device
03-24 08:40:48.398 5650-5650/com.github.mjdev.usbfileman E/MainActivity: error setting up device
                                                                         java.io.IOException: reading failed!
                                                                             at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:149)
                                                                             at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.java:82)
                                                                             at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.java:311)
                                                                             at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.java:267)
                                                                             at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:380)
                                                                             at com.github.mjdev.libaums.usbfileman.MainActivity.access$000(MainActivity.java:74)
                                                                             at com.github.mjdev.libaums.usbfileman.MainActivity$1.onReceive(MainActivity.java:93)
                                                                             at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:881)
                                                                             at android.os.Handler.handleCallback(Handler.java:739)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                             at android.os.Looper.loop(Looper.java:148)
                                                                             at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                                             at java.lang.reflect.Method.invoke(Native Method)
                                                                             at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                                             at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
03-24 08:40:48.399 5650-5650/com.github.mjdev.usbfileman I/Choreographer: Skipped 1260 frames!  The application may be doing too much work on its main thread.
03-24 08:40:48.469 5650-5673/com.github.mjdev.usbfileman I/Adreno: QUALCOMM build                   : 63c06b2, I8366cd0437
                                                                   Build Date                       : 10/21/15
                                                                   OpenGL ES Shader Compiler Version: XE031.05.13.02
                                                                   Local Branch                     : 
                                                                   Remote Branch                    : quic/LA.BF64.1.2.9_v2
                                                                   Remote Branch                    : NONE
                                                                   Reconstruct Branch               : NOTHING
03-24 08:40:48.482 5650-5673/com.github.mjdev.usbfileman I/OpenGLRenderer: Initialized EGL, version 1.4

mjdev, 2016-03-24:
Hey @greddyatpt okay, sounds strange. I do not have an Android M device yet so I test everything with Genymotion, and there it works fine for me. I will investigate on this asap.

Sure, my mail is: jahnen@in.tum.de

greddyatpt, 2016-03-24:
The issue could be to do with my USB 3.0 drive.  Let me do some more testing and add details.

greddyatpt, 2016-03-25:
It's working even with a USB 3.1 drive now.  **Please close this bug as "already fixed" or "not a bug."**  I was seeing some old (pre bug#10 fix) source files in the Studio.  When I deleted the .git folder, it picked up the latest files.  Probably to do with ignoring the VCS setup when I imported the project. Sorry for the false alarm.

mjdev, 2016-03-25:
All right, I am glad to hear that it works now :)

closed 2016-03-25 by mjdev