magnusja/libaums #21: Writing to ScsiBlockDevice failed

jmue, 2016-03-24:
I added a new android module implementing SAF Document Provider @ https://github.com/jmue/libaums/tree/feature/storageprovider. If I run my sample app `usbstorageprovider` @ https://github.com/jmue/libaums/tree/test I can browse my usb device. However, if I add this module to your sample app I get the following error:

`03-24 20:51:06.162 5010-5815/com.github.mjdev.usbfileman E/ScsiBlockDevice: Writing all bytes on command ScsiRead10 [blockAddress=574, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024] failed!
03-24 20:51:06.164 5010-5815/com.github.mjdev.usbfileman W/DocumentsProvider: Failed during query
                                                                              java.io.FileNotFoundException: reading failed!
                                                                                  at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider.queryChildDocuments(UsbDocumentProvider.java:177)
                                                                                  at android.provider.DocumentsProvider.query(DocumentsProvider.java:497)
                                                                                  at android.content.ContentProvider.query(ContentProvider.java:966)
                                                                                  at android.content.ContentProvider$Transport.query(ContentProvider.java:211)
                                                                                  at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:112)
                                                                                  at android.os.Binder.execTransact(Binder.java:446)
`

mjdev, 2016-03-25:
Hey, I had a look into your repository, but it seems that your sample application is empty?

Seems like writing to your USB thumb drive failed, but I cannot say why. Maybe you can provide some more details on what your are doing exactly, esp. what do you mean by "adding"? Only adding a module should not break the sample application.

mjdev, 2016-03-25:
Okay I had a second look over it and now I think I get the thing with the empty application. You application solely tests the StorageProvider, right? 

Strange that this does not work with the sample application. Unfortunately I can only test with the simulator at the moment and it seems that the simulator does not support Intent. ACTION_OPEN_DOCUMENT, which application are you using to test the StorageProvider itself?

mjdev, 2016-03-25:
Okay after some additional testing I can reproduce this error. The root directory is displayed perfectly, but after clicking on a folder this exception occurs:

```
03-25 08:44:23.587 3416-3998/com.github.mjdev.usbfileman D/UsbDocumentProvider: queryRoots()
03-25 08:44:23.587 3416-3998/com.github.mjdev.usbfileman D/UsbDocumentProvider: add root 31504014:
03-25 08:44:31.566 3416-3998/com.github.mjdev.usbfileman D/UsbDocumentProvider: queryDocument() 31504014:
03-25 08:44:31.566 3416-3998/com.github.mjdev.usbfileman D/UsbDocumentProvider: getFileForDocId() 31504014:
03-25 08:44:31.636 3416-3998/com.github.mjdev.usbfileman D/UsbDocumentProvider: queryChildDocuments() 31504014:
03-25 08:44:31.636 3416-3998/com.github.mjdev.usbfileman D/UsbDocumentProvider: getFileForDocId() 31504014:
03-25 08:44:39.046 3416-3427/com.github.mjdev.usbfileman D/UsbDocumentProvider: queryChildDocuments() 31504014:/test3
03-25 08:44:39.046 3416-3427/com.github.mjdev.usbfileman D/UsbDocumentProvider: getFileForDocId() 31504014:/test3
03-25 08:44:39.047 3416-3427/com.github.mjdev.usbfileman E/ScsiBlockDevice: Writing all bytes on command ScsiRead10 [blockAddress=92, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024] failed!
03-25 08:44:39.057 3416-3427/com.github.mjdev.usbfileman W/DocumentsProvider: Failed during query
                                                                              java.io.FileNotFoundException: reading failed!
                                                                                  at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider.queryChildDocuments(UsbDocumentProvider.java:176)
                                                                                  at android.provider.DocumentsProvider.query(DocumentsProvider.java:498)
                                                                                  at android.content.ContentProvider.query(ContentProvider.java:1017)
                                                                                  at android.content.ContentProvider$Transport.query(ContentProvider.java:238)
                                                                                  at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:112)
                                                                                  at android.os.Binder.execTransact(Binder.java:453)
```

Maybe this has something to do that usbfileman is running in the background and also has an open connection to the USB device? Is that the case when you get the exception, too?

jmue, 2016-03-25:
Yes, my sample solely tests the storageprovider. And yes, usbfileman is running in the background in the error case, this is probably the cause. I don't know if this scenario is technically possible, how is your http Server acting?

jmue, 2016-03-25:
Opening a pull request once this is working was my intention. ☺

mjdev, 2016-03-25:
The HTTP server is not running by default. 

I also tried the following:

````
if (device != null) {
			Log.d(TAG, "Closing device");

			device.close();
		}

		// ACTION_OPEN_DOCUMENT is the intent to choose a file via the system's file
		// browser.
		Intent intent = new Intent(Intent.ACTION_OPEN_DOCUMENT);
		intent.setType("*/*");

		startActivityForResult(intent, 42);
```

Unfortunately with the same result.

mjdev, 2016-05-08:
Hey @jmue can you maybe nevertheless open up a pull request?

jmue, 2016-05-10:
Ok, here we go.

mjdev, 2016-05-17:
Thank you! Will be in v0.4 ;)

mjdev, 2016-06-23:
Hey @jmue 

I player around today, and noticed if I close the example application, then the storage provider works and I can switch to different sub directories without the strange writing error.

46472761, 2017-04-06:
JAMES LACONIA DAVIS,TRIPLE ELITE,46472761