magnusja/libaums #23: Sample app crashes on Android 6.0.1

tral, 2016-04-12:
Sample app crashes on Android 6.0.1 when I attach USB device.
Nexus 5 Stock

mjdev, 2016-04-12:
Can you please provide some logs/stack traces?

tral, 2016-04-12:
no, I'm just trying to use this app as end-user and it simply crashes on Nexus 5 with stock Android 6.0.1

mjdev, 2016-04-12:
Okay, how did you install it on you phone without some dev environment?

tral, 2016-04-12:
just opened in android studio and launched )

Shuhrat-java, 2016-04-13:
@tral I have Nexus 7 Android 6.0.1 and I also just opened in android studio and launched. However I did not have any crash. So you should put logs which caused crash. For that you should set wifi debugging. 
Steps:
adb kill-server
adb tcpip 5555
adb connect "your device ipaddress in same network with your device, ex: 192.168.0.100"

tral, 2016-04-13:
`FATAL EXCEPTION: main
                                                                             Process: com.github.mjdev.usbfileman, PID: 19505
                                                                             java.lang.RuntimeException: Error receiving broadcast Intent { act=com.github.mjdev.libaums.USB_PERMISSION flg=0x10 (has extras) } in com.github.mjdev.libaums.usbfileman.MainActivity$1@e41635c
                                                                                 at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:891)
                                                                                 at android.os.Handler.handleCallback(Handler.java:739)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                 at android.os.Looper.loop(Looper.java:148)
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
                                                                                 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
                                                                              Caused by: java.lang.IndexOutOfBoundsException: Invalid index 0, size is 0
                                                                                 at java.util.ArrayList.throwIndexOutOfBoundsException(ArrayList.java:255)
                                                                                 at java.util.ArrayList.get(ArrayList.java:308)
                                                                                 at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:424)
                                                                                 at com.github.mjdev.libaums.usbfileman.MainActivity.access$000(MainActivity.java:79)
                                                                                 at com.github.mjdev.libaums.usbfileman.MainActivity$1.onReceive(MainActivity.java:98)
                                                                                 at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:881)
                                                                                 at android.os.Handler.handleCallback(Handler.java:739) 
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95) 
                                                                                 at android.os.Looper.loop(Looper.java:148) 
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:5417) 
                                                                                 at java.lang.reflect.Method.invoke(Native Method) 
                                                                                 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) `

mjdev, 2016-04-13:
Strange, seems like libaums cannot find any partition on the disk.

Please make sure the usb flash drive is formatted with MBR and a FAT32 partition. GUID (often used in OS X) and NTFS partitions (Windows) are not supported.

tral, 2016-04-14:
I tried with FAT32 (32GB) flash-drive and with Blood Glucose Monitor with FAT16 (800Kb)

mjdev, 2016-04-16:
What is Blood Glucose Monitor?

Can you provide me with an image of the stick, so I can try it myself?

mjdev, 2016-05-08:
Closed due to inactivity, if you encounter the error again, please reopen.

closed 2016-05-08 by mjdev

Y-E-P, 2016-10-25:
I know about this bug and I fixed it.The bug in MainActivity,when we get unsupported partition and null file system.