magnusja/libaums #24: Move fails for file which is under the root folder

diogowbrito, 2016-04-19:
`java.lang.RuntimeException: An error occurred while executing doInBackground()
                                                                        at android.os.AsyncTask$3.done(AsyncTask.java:309)
                                                                        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
                                                                        at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
                                                                        at java.util.concurrent.FutureTask.run(FutureTask.java:242)
                                                                        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
                                                                        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                        at java.lang.Thread.run(Thread.java:818)
                                                                     Caused by: java.lang.NullPointerException: Attempt to invoke direct method 'void com.github.mjdev.libaums.fs.fat32.FatDirectory.init()' on a null object reference
                                                                        at com.github.mjdev.libaums.fs.fat32.FatDirectory.move(FatDirectory.java:612)
                                                                        at com.github.mjdev.libaums.fs.fat32.FatFile.moveTo(FatFile.java:200)`

mjdev, 2016-04-22:
Can you maybe provide some additional information regarding the move.

diogowbrito, 2016-04-22:
The operation crashes when moving a file which is under the root to another folder two levels under the root.
Crashes when performing this move operation
Root > File
Root > FolderA > File

mjdev, 2016-05-08:
Can you check current develop if this issue is resolved?

diogowbrito, 2016-05-09:
@mjdev looks like it's solved 👍

mjdev, 2016-05-09:
Very nice!

closed 2016-05-09 by mjdev