magnusja/libaums #25: Occasional crash while reading USB file content

greddyatpt, 2016-04-21:
App crashes occasionally with the the following stacktrace.  Environment: Samsung S7 and Moto X Play phones running Android Marshmallow 6.0.1.  USB is Sandisk dual drive.

04­19 16:31:34.189 29157 29157 D UsbDeviceConnectionJNI: close
04­19 16:31:34.189 3445 3445 D EdgeLight: Turning off
04­19 16:31:34.189 29157 30411 E NanoHTTPD: Could not send response to the client 
04­19 16:31:34.189 29157 30411 E NanoHTTPD: java.io.IOException: reading failed!
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:149)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:215)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at com.github.mjdev.libaums.partition.Partition.read(Partition.java:124)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at com.github.mjdev.libaums.fs.fat32.ClusterChain.read(ClusterChain.java:94)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.java:159)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.java:104)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.sendBody(NanoHTTPD.java:1476)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectEncoding(NanoHTTPD.java:1453)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectTransferAndEncoding(NanoHTTPD.java:1443)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.send(NanoHTTPD.java:1429)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:852)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:189)
04­19 16:31:34.189 29157 30411 E NanoHTTPD: at java.lang.Thread.run(Thread.java:818)
04­19 16:31:34.189 29157 30430 E UsbDeviceConnectionJNI: device is closed in native_control_request
04­19 16:31:34.189 29157 30430 E ScsiBlockDevice: Writing all bytes on command ScsiRead10 [blockAddress=12048, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024] failed!
04­19 16:31:34.189 29157 30430 E UsbDeviceConnectionJNI: device is closed in native_control_request
04­19 16:31:34.189 29157 30430 E NanoHTTPD: Could not send response to the client 
04­19 16:31:34.189 29157 30430 E NanoHTTPD: java.io.IOException: reading failed!
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:149)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:215)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.partition.Partition.read(Partition.java:133)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:134)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:63)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.fs.fat32.FatFile.initChain(FatFile.java:89)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.java:157)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.java:104)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.sendBody(NanoHTTPD.java:1476)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectEncoding(NanoHTTPD.java:1453)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.sendBodyWithCorrectTransferAndEncoding(NanoHTTPD.java:1443)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$Response.send(NanoHTTPD.java:1429)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:852)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:189)
04­19 16:31:34.189 29157 30430 E NanoHTTPD: at java.lang.Thread.run(Thread.java:818)

mjdev, 2016-04-22:
`04­19 16:31:34.189 29157 29157 D UsbDeviceConnectionJNI: close`

What are you doing exactly that the device gets closed?

greddyatpt, 2016-04-25:
Sorry, this is one of those rare crashes.  Haven't been able to reproduce it yet to answer your question.  Please feel free to close it for now.  If the crash happens again, I'll enter a bug with more details.

closed 2016-05-08 by mjdev