magnusja/libaums #26: createDirectory deletes all other files in directory

PNowell67, 2016-04-22:
Steps to reproduce
1. getRootDirectory().search(path) to find a UsbFile
2. call createDirectory on returned UsbFile
result
the directory is created but all other files are deleted

the problem is that the UsbFile returned by search is not fully initialised and the 'entries' field is
zero even though the directory does have entries.

Workaround is to call list() or listFIles() after search(path) which will init the UsbFile

Solution could be to ensure that UsbFile returned from search(path) is fully
initialised or init() is called in all methods such as createDirectory and createFile

mjdev, 2016-04-22:
Thanks, I think initializing in in search is a good option. If you have time, a pull request is appreciated.

mjdev, 2016-04-22:
Okay actually I think it is better to do it in createDirectory, for performance reasons.

PNowell67, 2016-04-22:
Yes, in both createDirectory and createFile
I think it would also be better to assign the entries field in init() rather than the constructor,
then, if there are any other problems, there would be a null pointer exception instead of loss of data.

Can you give me instructions on how to do pull request, not obvious from github.

mjdev, 2016-04-25:
Sure, just fork my repository, then clone it do your change commit and push the change and then in the github web UI you can find "file a pull request" or something like that on your forked repo.

closed 2016-05-08 by mjdev