magnusja/libaums #29: Unfortunately, Usbfileman has stopped

norani, 2016-05-17:
`
05-17 12:08:18.481 20116-20116/com.github.mjdev.usbfileman I/ScsiBlockDevice: Block size: 512
05-17 12:08:18.481 20116-20116/com.github.mjdev.usbfileman I/ScsiBlockDevice: Last block address: 15556607
05-17 12:08:18.496 20116-20116/com.github.mjdev.usbfileman W/Partition: unsupported partition type: 110
05-17 12:08:18.496 20116-20116/com.github.mjdev.usbfileman W/Partition: unsupported partition type: 116
05-17 12:08:18.496 20116-20116/com.github.mjdev.usbfileman D/AndroidRuntime: Shutting down VM
05-17 12:08:18.501 20116-20116/com.github.mjdev.usbfileman E/AndroidRuntime: FATAL EXCEPTION: main
                                                                             Process: com.github.mjdev.usbfileman, PID: 20116
                                                                             java.lang.RuntimeException: Error receiving broadcast Intent { act=com.github.mjdev.libaums.USB_PERMISSION flg=0x10 (has extras) } in com.github.mjdev.libaums.usbfileman.MainActivity$1@28a6c7ab
                                                                                 at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:933)
                                                                                 at android.os.Handler.handleCallback(Handler.java:739)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                 at android.os.Looper.loop(Looper.java:145)
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:5942)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
                                                                                 at java.lang.reflect.Method.invoke(Method.java:372)
                                                                                 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1400)
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1195)
                                                                              Caused by: java.lang.IndexOutOfBoundsException: Invalid index 1, size is 0
                                                                                 at java.util.ArrayList.throwIndexOutOfBoundsException(ArrayList.java:255)
                                                                                 at java.util.ArrayList.get(ArrayList.java:308)
                                                                                 at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:424)
                                                                                 at com.github.mjdev.libaums.usbfileman.MainActivity.access$000(MainActivity.java:79)
                                                                                 at com.github.mjdev.libaums.usbfileman.MainActivity$1.onReceive(MainActivity.java:98)
                                                                                 at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:923)
                                                                                 at android.os.Handler.handleCallback(Handler.java:739) 
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95) 
                                                                                 at android.os.Looper.loop(Looper.java:145) 
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:5942) 
                                                                                 at java.lang.reflect.Method.invoke(Native Method) 
                                                                                 at java.lang.reflect.Method.invoke(Method.java:372) 
                                                                                 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1400) 
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1195) 
05-17 12:08:18.551 20116-20116/com.github.mjdev.usbfileman I/Process: Sending signal. PID: 20116 SIG: 9
`

Above is my logcat. I think the error is caused by me changing from API level 23 to level 22.

mjdev, 2016-05-17:
The log says:
```
05-17 12:08:18.496 20116-20116/com.github.mjdev.usbfileman W/Partition: unsupported partition type: 110
05-17 12:08:18.496 20116-20116/com.github.mjdev.usbfileman W/Partition: unsupported partition type: 116
```

So changing the API level should not be the problem. Does it work when using API level 23?

What filesystems are on the flash drive? I cannot find any information for the types given in the log on https://en.wikipedia.org/wiki/Partition_type

norani, 2016-05-18:
If developer uses api level 23, the app will crash when trying to run the app if not put some code to handle the runtime permission (See https://inthecheesefactory.com/blog/things-you-need-to-know-about-android-m-permission-developer-edition/en).

My USB file system is FAT32. Last two weeks, I run libaums app in my phone and it can run perfectly. Can create file, rename and etc. But yesterday, it crash. I already fork new one and delete the new one but still crash

closed 2016-05-20 by norani

mjdev, 2016-05-20:
Hey @norani, did you somehow solve the problem?