magnusja/libaums #30: How to read hidden area of USB?

norani, 2016-05-20:
Hi,

I use libaums to access private partition in usb mass storage.  Then I want to access hidden partition in usb mass storage. I have the vendor specific opcode to read and write hidden entry. Firstly, how to read the capacity of hidden area and the block size, also last block address? Is there something to do with `private static final int FLAG_HIDDEN = 0x02;` and ```
boolean isHidden() {
		return isFlagSet(FLAG_HIDDEN);
	} ? Or do I need to create new class like ScsiReadHidden like ScsiRead10 class?
```

mjdev, 2016-05-20:
Hey @norani,

can you maybe give some more information about hidden partitions? Are you referring to the secured FAT16 partition (116)?

Is a hidden partition a usual FAT32 partition? Or somehow encrypted?

norani, 2016-05-21:
I'm sorry, actually hidden is not a partition. I read hidden area of USB by hard code below 
` read hidden
				ByteBuffer cbwBuffer = ByteBuffer.allocate(31); // send ReadHidden command
				byte[] cswBuffer= new byte[512];// for Read data IN
				Arrays.fill(cswBuffer, (char) 0);// put ALL 0 first

				cbwBuffer.put(0,(byte)0x55);
				cbwBuffer.put(1,(byte)0x53);
				cbwBuffer.put(2,(byte)0x42);
				cbwBuffer.put(3,(byte)0x43);

				cbwBuffer.put(4,(byte)0x0);
				cbwBuffer.put(5,(byte)0x0);
				cbwBuffer.put(6,(byte)0x0);
				cbwBuffer.put(7,(byte)0x0);

				cbwBuffer.put(8,(byte)0x0);
				cbwBuffer.put(9,(byte)0x02);
				cbwBuffer.put(10,(byte)0x0);
				cbwBuffer.put(11,(byte)0x0);

				cbwBuffer.put(12,(byte)0x80);
				cbwBuffer.put(13,(byte)0x0);
				cbwBuffer.put(14,(byte)0x10);

				cbwBuffer.put(15,(byte)XXXX); //vendor specific opcode to read hidden

				cbwBuffer.put(16,(byte)0x0);

				cbwBuffer.put(17,(byte)0x0);
				cbwBuffer.put(18,(byte)0x0);
				cbwBuffer.put(19,(byte)0x0);
				cbwBuffer.put(20,(byte)0x1); //request 1 sector

				cbwBuffer.put(21,(byte)0x0);

				cbwBuffer.put(22,(byte)0x0);
				cbwBuffer.put(23,(byte)0x1); //start sector number to be read

				int iRes = conn.bulkTransfer(epOUT, cbwBuffer.array() , 31, 2000);
				iRes = conn.bulkTransfer(epIN, cswBuffer, 512, 2000); //receive hidden area` Instead using this hard code, can I just change/modified certain code from libaums ?

mjdev, 2016-05-23:
I would suggest to subclass `CommandBlockWrapper`like the `ScsiInquiry` command does.

mjdev, 2016-06-08:
I am closing this now, if you need further assistance, please reopen.

closed 2016-06-08 by mjdev