magnusja/libaums #32: ZipInputStream from UsbFileInputStream fails with BufferOverflowException

diogowbrito, 2016-06-09:
I'm trying to decompress a file in the drive. For such I am doing something similar to this:

```
UsbFile zippedFile = getFile(path);
ZipInputStream zipInputStream = new ZipInputStream(new UsbFileInputStream(zippedFile));
ZipEntry entry;
while ((entry = zipInputStream.getNextEntry()) != null) {
   …
}
…
```
However, this code fails when `getNextEntry()` is called. I've given it a look at it seems to be related to the UsbFileInputStream implementation, although I am not 100% sure what is causing such behavior.

mjdev, 2016-06-09:
Hey @diogowbrito,

thanks for the feedback, do you habe some exception stack trace or similar?

diogowbrito, 2016-06-09:
Yeah, my bad. At the time the app was simply closing and I wasn't being able to get one, but finally have it. Here you go:

```
Caused by: java.nio.BufferOverflowException
   at java.nio.ByteBuffer.put(ByteBuffer.java:787)
   at com.github.mjdev.libaums.partition.Partition.read(Partition.java:127)
   at com.github.mjdev.libaums.fs.fat32.ClusterChain.read(ClusterChain.java:94)
   at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.java:159)
   at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.java:104)
   at java.io.PushbackInputStream.read(PushbackInputStream.java:185)
   at libcore.io.Streams.readFully(Streams.java:81)
   at java.util.zip.ZipInputStream.getNextEntry(ZipInputStream.java:246)
```

mjdev, 2016-07-18:
Can you maybe upload a minimal example where this is happening, so I can easily try it on my machine? Thanks!

magnusja, 2016-12-16:
Hey I think I finally fixed this one. Please have a look on develop.

closed 2016-12-16 by magnusja