magnusja/libaums #33: Create folder operation successful but after unplug/plug device folder is not there

diogowbrito, 2016-06-28:
I had noticed this behaviour at least since 3.0 but after pulling the latest code (5dfe700) it is still noticeable.

Steps to reproduce (even with the given test libaums app) are:

1. Under any folder in root create a subfolder.
2. The operation will go ok and the folder will be there.
3. Unplug and plug the device.
4. Navigate into that folder. The created folder is not there.

mjdev, 2016-07-18:
Unfortunately I am not able to reproduce this currently.

Which Android version and which device are you using?
Does this also happen if you use the genymotion simulator?

rameshvoltella, 2016-12-16:
@magnusja  ya am also getting this bug as @diogowbrito say .If we create a folder in main root directory it will not disappear even  we unplug the usb, But if we create a folder inside any folder in the root directory it will disappear once we un plug the usb.if we create a file it will not disappear this issue occur only if we create a folder under any folder in root directory.(The created folder only disapper after we unplug the usb)

Device info am testing :
Samsung Galaxy On5 Pro
Version-6.0
Model No-SM-G550FY
Some screen shots
Screen 1
Directory in root file -

[image]

Screen 2
Created a subfolder and file inside folder **folderinroot**

[image]

Screen3
After creating the folder and file i un-pluged the device and plugged back but now when i go to the **folderinroot** folder(name of folder in the root directory), i cant see the folder i created only the file is visble the folder created has disappear

[image]

What may be the reason for the disappearance of the folder i cant find anything on debugging ?, i think this is a high priority bug.

magnusja, 2016-12-16:
Changes are written immediately to disk (https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L382).
I quickly tested it with my example app, and did not have a problem. Do you have the same problem with my example?

Do you see the USB drive blinking when creating a directory? Do you see that there is actually data written in the logs, resp. if you debug?

How long do you wait until detaching the pen drive?

rameshvoltella, 2016-12-16:
@magnusja did you created a subfolder inside a folder in your example

rameshvoltella, 2016-12-17:
any solution?

rameshvoltella, 2016-12-19:
@magnusja  any solution?

magnusja, 2016-12-19:
@rameshvoltella 
Unfortunately, I have no news. The only thing I find a little weird is that it says it has to round. But usually the file system stuff should always be aligned with the block size.

Can you tell me the block size of the drive and the cluster size of the FAT32?

Maybe you can uncomment the logs in Partition#write and ScsiBlockDevice#write.

rameshvoltella, 2016-12-20:
I/ScsiBlockDevice: Block size: 512

When create folder in sub folder

> 
> 12-20 12:21:45.381 9740-9740/com.filemangerapp.anything D/ViewRootImpl: ViewPostImeInputStage processPointer 0
> 12-20 12:21:45.491 9740-9740/com.filemangerapp.anything D/ViewRootImpl: ViewPostImeInputStage processPointer 1
> 12-20 12:21:45.501 9740-9740/com.filemangerapp.anything D/ScsiBlockDevice: writing: ScsiWrite10 [blockAddress=2122, transferBytes=1024, blockSize=512, transferBlocks=2, getdCbwDataTransferLength()=1024]
> 12-20 12:21:45.501 9740-9740/com.filemangerapp.anything D/FsInfoStructure: writing to device
> 12-20 12:21:45.501 9740-9740/com.filemangerapp.anything D/ScsiBlockDevice: writing: ScsiWrite10 [blockAddress=2049, transferBytes=512, blockSize=512, transferBlocks=1, getdCbwDataTransferLength()=512]
> 12-20 12:21:45.506 9740-9740/com.filemangerapp.anything I/FAT: allocating clusters finished
> 12-20 12:21:45.506 9740-9740/com.filemangerapp.anything D/FatDirectory: adding entry: [FatLfnDirectoryEntry getName()=koop] with short name: KOOP
> 12-20 12:21:45.506 9740-9740/com.filemangerapp.anything W/ScsiBlockDevice: we have to round up size to next block sector
> 12-20 12:21:45.506 9740-9740/com.filemangerapp.anything D/ScsiBlockDevice: writing: ScsiWrite10 [blockAddress=6464, transferBytes=512, blockSize=512, transferBlocks=1, getdCbwDataTransferLength()=512]
> 12-20 12:21:45.511 9740-9740/com.filemangerapp.anything W/ScsiBlockDevice: we have to round up size to next block sector
> 12-20 12:21:45.511 9740-9740/com.filemangerapp.anything D/ScsiBlockDevice: writing: ScsiWrite10 [blockAddress=15360, transferBytes=512, blockSize=512, transferBlocks=1, getdCbwDataTransferLength()=512]

magnusja, 2016-12-21:
Cluster size?

rameshvoltella, 2016-12-21:
I dont find the lig for that

magnusja, 2016-12-21:
FileSystem.getChunkSize()

rameshvoltella, 2016-12-21:
Wait for 10 min let me turn my laptop on once again where should i put the log to get the cluster size is that on fat. 32 class?

magnusja, 2016-12-21:
Nah it's Partition.getFileSystem and then you can access that anywhere in your code.

rameshvoltella, 2016-12-21:
i dont get the methord getChunkSize()

magnusja, 2016-12-21:
Are you using current develop branch?

rameshvoltella, 2016-12-21:
yes your sample project let me recheck

magnusja, 2016-12-21:
Should be there then.
```java
Log.d(TAG, "Chunk size: " + currentFs.getChunkSize());
```

https://github.com/magnusja/libaums/blob/develop/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.java#L574

rameshvoltella, 2016-12-21:
Chunk size: 32768

magnusja, 2016-12-21:
Is it a 128GB stick?

rameshvoltella, 2016-12-21:
no its jut 4 gb pendrive

magnusja, 2016-12-21:
Okay interesting, I will format a stick of mine with 32K cluster size and then lets see what happens

rameshvoltella, 2016-12-21:
i tested in samsung device as i mentioned above it failed, but in moto device its working, do you think its issue with the pendrive or the devices. i only tested in samsung and moto device

rameshvoltella, 2016-12-21:
@magnusja r u testing now?

magnusja, 2016-12-21:
Yes I did a test.

```
Fat32BootSector{bytesPerSector=512, sectorsPerCluster=64, reservedSectors=32, fatCount=2, totalNumberOfSectors=30713472, sectorsPerFat=3749, rootDirStartCluster=2, fsInfoStartSector=1, fatMirrored=true, validFat=0, volumeLabel=''}
```

I am still unable to reproduce. Do you have the problem also with my example app?

I think it might be a problem with your device. Data is definitely written to the USB drive. Maybe you the Motorola Android has buffering issues. Do you just plug it out or do you also wait until the call to ```UsbDevice.close()``` succeeded?

rameshvoltella, 2016-12-21:
i never un plug the device, the close call is a sucess because the samsung mount again after i close or example application. i have never unplug the device, also the issue is with samsung device not motorola

magnusja, 2016-12-21:
I see. Can you maybe deactivate the mounting of the Samsung device? Maybe that interferes somehow with my library.

rameshvoltella, 2016-12-21:
@magnusja  i dont find any option to stop that, 
Device info am testing :
Samsung Galaxy On5 Pro
Version-6.0
Model No-SM-G550FY

you can easily reproduce this issue if you use this model

i can make a video how this is happening if you want to see?

rameshvoltella, 2016-12-21:
@magnusja can you tell me what difference between when a file and folder creation because the files created in subfolder of root directory is not disappearing, only folder is disappearing.
what does **dotdot entry** do when create a sub folder?

magnusja, 2016-12-21:
Yeah I don't have that model and I don't think that I am going to buy it :D
Nah, I belive you.

Okay that is interesting. The content of a file can be any bytes you wish. The content of directories is in a certain format: fat directory entries. Each entry is 32 bytes and describes either another sub dir or a file. 

The dot and the dotdot entries are special entries every directory has (except root dir). Dot entry is a link to the current directory you are in. dotdot is a link to the parent directory.

You can find a detailed description in my thesis.

rameshvoltella, 2016-12-21:
@magnusja let me give a try :)

magnusja, 2016-12-23:
Hey @rameshvoltella,

instead of the synchronous UsbDeviceCommunication I added a feature to switch to the asynchronous UsbRequest to transfer bulk. You can find it here: https://github.com/magnusja/libaums/tree/feature/usb-request

Before initializing any device, you have to specify the UsbCommunication method:

```
UsbCommunicationFactory.setUnderlyingUsbCommunication(UnderlyingUsbCommunication.USB_REQUEST_ASYNC);
```

Maybe that helps, worth a try.

rameshvoltella, 2016-12-23:
@magnusja  

i did as per you say but still issue remains

> UsbManager usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
> 		// we only use the first device
> 		UsbCommunicationFactory.setUnderlyingUsbCommunication(UsbCommunicationFactory.UnderlyingUsbCommunication.USB_REQUEST_ASYNC);
> 
> 		UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(this);

is the methord is what you mentionerd

magnusja, 2016-12-23:
No looks good. Do you see `D/UsbRequestJNI: init` in the logs?

rameshvoltella, 2016-12-24:
@magnusja  first HAPPY XAMS :)

yes i am getting the log

> 12-25 00:57:50.305 8730-8730/com.github.mjdev.usbfileman D/UsbMassStorageDevice: setup device
> 12-25 00:57:50.320 8730-8730/com.github.mjdev.usbfileman D/UsbRequestJNI: init

magnusja, 2017-01-05:
Hey,

happy new year.
No, I don't have any additional ideas to try. Did you find something out during debugging?

46472761, 2017-04-06:
JAMES LACONIA DAVIS,TRIPLE ELITE,46472761

howlingkaze, 2019-02-21:
@magnusja 
I am facing exactly the same issue.
After trace code. I've noticed that you've made a comment in FAT.java line 212.
// TODO we should write in in all FATs when they are mirrored!

So i do write the same data to other mirrored FAT, and it's seems to solve this issue.
Folders remain exist after i unplug and plug again.

magnusja, 2019-02-21:
Hey @howlingkaze , thanks a lot for your feedback. This is very interesting news. Are you 100% sure that it works? It seems a little odd to me because usually one FAT is sufficient, but maybe there is a bug in figuring out the right one. Can you maybe file a PR so I can include this fix into the mainline? Thanks

howlingkaze, 2019-02-21:
@magnusja 
I've made a mistake, after i mark my code, the create folder isn't disappear after unplug either.
My test code is clone from dev branch, so its seem that some commit in dev-branch fix this issue, but i dunno which one does.

Sorry for the disturbance.