magnusja/libaums #34: After renaming a file the modified date is not updated. Dates seems incorrect.

diogowbrito, 2016-07-11:
After calling `renamedFile.setName` the file modified date is not updated.

Oddly, I found two other issues.
1) `lastModified` is older than `createdAt`,
2) The returned values are different in between different calls (values from debug).

```
renamedFile.lastModified() = 1467586800661
renamedFile.lastModified() = 1467586800487
renamedFile.createdAt()    = 1467632242549
renamedFile.createdAt()    = 1467632242909
```

mjdev, 2016-07-11:
1. Hmm not sure whether renaming should alter the modified date.
2. Is that also the case before renaming?

diogowbrito, 2016-07-11:
1. In fact that is a good opinion. I just based my knowledge on some cloud accounts and they seem to update it but mac os, for instance, does not.
2. Double checked with a different file before any change and it has the same behaviour - lastModified older than createdAt and values change in between calls.

```
originalFile.createdAt()    = 1467887186742
originalFile.lastModified() = 1467846000878
originalFile.createdAt()    = 1467887186103
originalFile.lastModified() = 1467846000326
```

mjdev, 2016-07-18:
Please try my fix, somehow modified and last accessed got confused.

closed 2016-09-20 by magnusja