magnusja/libaums #36: USBFile .close/flush updates the memory of its parent Dir , ignoring the changes happened to that Dir post file creation

aravind14feb, 2016-09-30:
I faced this problem when I had to download a list of files on the USB.

Initially i created 10 empty UsbFiles on the USB drive and inserted them into a hashmap with key as the downloadUrl.
(lets say the files created are test1.txt, test2.txt .... test10.txt)

and now to write the downloaded data into the files, i iterate the map and get the UsbFile for the corresponding downloadUrl. consider the order of files that is being written is as shown below,

test8.txt,
test10.txt
...

test1.txt.

After writing all the files, if i goto a file explorer and check the Usb storage, all i could see is only test1.txt. (the file which was written at the last, and which was created at first)

I guess the reason for this problem, when the "parent" of the UsbFile is being updated, via. usbfile.flush(), the memory associated with the corresponding instance of the "parent" is considered as the latest content of the "parent", ignoring all the other changes happened after usbfile creation. (when test1 was created, its parent had only one child and which was retained even when the file was written at last)

On debugging, i was able to see this
map.get(downloadUrl_of_test1.txt).getParent().listFiles() will have only 1 child.
map.get(downloadUrl_of_test2.txt).getParent().listFiles() will have 2 children

map.get(downloadUrl_of_test8.txt).getParent().listFiles() will have 8 children

Since the test1.txt is written after all other files, and  its parent directory contains only 1 child, at the end of downloading files, we are able to see only test1.txt (as the write is being done on Parent level, based on what i understood fromt he comments)

                // we only have to update the parent because we are always writing
		// everything
		// immediately to the device
		// the parent directory is responsible for updating the
		// FatDirectoryEntry which
		// contains things like the file size and the date time fields

magnusja, 2016-10-03:
Hey, thanks for your feedback.

How do you create these files? Usually, I would assume that every file has the same parent directory object instance.

aravind14feb, 2016-10-04:
Hey Magnus,

First of all, thanks for the library and for your contribution.. Been really helpful..

We create the files in a method from a loop, to which we pass the RootDirectory of the UsbDrive, and a relative path in which the file has to be created.. 

Inside the function, we breakdown the relative path and search for each of the directory or the file to exist in the specific order, if it doesnt exist, we go ahead and create them.

magnusja, 2016-10-04:
Can you paste a little snippet how you do that?

magnusja, 2016-10-13:
Hey @aravind14feb 
is this still an issue?

aravind14feb, 2016-10-14:
@magnusja  We have proceeded with a workaround for now as we are closing in a deadline. I will try to post the sample for the above problem soon. Sorry about the delay.

aravind14feb, 2016-10-24:
I was not able to reproduce the problem in the sample app. I will reopen the issue if i could capture the problem

closed 2016-10-24 by aravind14feb

magnusja, 2016-10-25:
Okay cool thank you!