magnusja/libaums #37: Junk files created randomly

aravind14feb, 2016-10-24:
@magnusja 

I'm not able to square out a scenario for this problem, but when I download files to the USB drive / copy files from Internal storage to Usb drive , I get some junk files randomly.

I have been using UsbFileOutputstream and UsbFileInputStream for writing / reading data to and from the UsbFiles.

Have u faced any such problems during the course of your development and

magnusja, 2016-10-25:
Which version are you using? 
This bug could be similar: https://github.com/magnusja/libaums/issues/14

aravind14feb, 2016-10-26:
I have been using the version 0.3.1.  I did see the #14 .  Even I have been doing several moveTo method calls.  
I will test more to check if it is related to moveTo calls or if it is a common case.

aravind14feb, 2016-10-27:
The problem occurs very randomly and i was able to capture one instance . In below given log, there are certain weird characters displayed as volume label and when i check my USB stick i see the junk files. Not sure if this will be of any use to you.

> D/FatDirectory: search entry: EData
 D/FatDirectory: search file: hex
 D/FatDirectory: search entry: hex
 D/FatDirectory: search file: Preferences.xml
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: VU￨V
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: p[ￎﾱp
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: {￷ﾾ|
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: ﾉￌￊ^ﾋ
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: ﾜﾟ
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: ﾽk1ﾽ￿￿
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: ￖoﾠￖ
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: ￜ-ￎAￜ
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: uNG
 W/FatDirectory: volume label in non root dir!
 D/FatDirectory: volume label: \VR
>

magnusja, 2016-11-01:
Hmm, this looks really weird. Unfortunately, currently my time is very limited, but I will try to look into at asap.

Maybe it would be useful to write some sort of test which randomly does some file operations to see when it occurs most.

aravind14feb, 2016-11-02:
I will try to write some test app which does the file writing and replacing stuff (got to do almost all of the operations :) )

One test case where the problem was reproduce-able is :
1. Download the files (like below) to the USB 
Folder A
   File 1
   File 2
   File 3
Folder B
  File 4
  File 5
  File 6
2. Re-download the same files to the USB
    To manage the proper sync, we had to 
     -> take a backup of Folder A, Folder B (say Folder A_bak, Folder B_bak) 
     -> download the contents onto the  Folder A_bak, Folder B_bak 
     -> after successful download, we will rename the original folders Folder A, Folder B as  Folder A_old,
     -> rename the Folder A_bak, Folder B_bak as Folder A , Folder B
    -> delete the _old folders

This being the usecase we generally test for this problem, we also get something like this 
[image]

after the initial download. (Windows always suggests some clean up of several entries that has ShortenedName like FIL1.M0 etc)

magnusja, 2016-11-02:
Can you maybe show how you use the API exactly? Maybe there is a threading problem. Are these sequential downloads or parallel?

aravind14feb, 2016-11-03:
@magnusja  its a sequential download. However I will post the sample with how we use the api.. will work on it today..

magnusja, 2016-11-10:
@aravind14feb Okay thanks. 
If you somehow manage it to give an example where I can reproduce this easily would be really helpful :)

aravind14feb, 2016-11-17:
@magnusja Sorry for the dealy.

I have drafted a sample app which does a transfer of files from local directory of the app to the USB. The sample app also contains several methods that we have used in our actual app to download / move files to USB.  The Junk file problem has been random and i was not able to reproduce it with this sample app so far. But could you take a look and let me know if I have used the apis of Libaums properly?
https://www.dropbox.com/s/aq5y1l006leb643/Sample1.zip?dl=0

Also we found the following stuff in our analysis
1. In certain cases, the junk files are actually the contents that we write into a file. (some times the name of the junk files are readable characters and some times byte data)

2.  Consider we have these file names in a directory which has to be moved to USB
DEBUG_LOG_161014195817.TXT
DEBUG_LOG_161014195817 - Copy.TXT
DEBUG_LOG_161014195817 - Copy (2).TXT

The shortfile names that were generated within libaums are not properly incremented. After the transfer if we plugin in the USB to

 a Windows machine, it suggests us to scan and clean the drive (since duplicate entries of DEBUG~1.TXT  are found in the drive) and on cleaning it, the files with the same shortname are renamed and the actual names of the file are updated with the modified short names. 
DEBUG~1.TX0
DEBUG~1.TX1
DEBUG~1.TX2

haoxi911, 2016-11-18:
@magnusja We also met the two issues: 1) randomly file corruption and hard to find a reproducible way, 1) the SFN issue when several files have same prefix and are saved into same folder. We do have a patch on the 2nd one, I will see if our Android developer can send a PR on the fix. Thanks!

aravind14feb, 2016-11-18:
@haoxi911 Thanks for your help. Btw, were you able to find any solution to the problem 1 ?

haoxi911, 2016-11-18:
@aravind14feb Not yet, I think we met the same issue that we can't find an easy way to reproduce it, but it does happen randomly.

aravind14feb, 2016-11-18:
@haoxi911 So far we just have faced corrupted files on the USB.  I'll wait for @magnusja to comment on this.

Meanwhile, your patch for the shortname entry will also be useful for me.

aravind14feb, 2016-11-18:
@haoxi911 Overriding the hashcode method in ShortName helped me to solve the problem 2.

magnusja, 2016-11-18:
Hey guys, thanks for your feedback!
A PR on problem 2 (Shortname) would be cool because I have very limited time currently.
I will play around and take a look into problem 2. Can you maybe tell me which phones you are using and which Android versions? Thanks!

haoxi911, 2016-11-19:
@aravind14feb It sounds like a better / easier fix than our approach, can you also share your approach? @magnusja I have included our changes in PR #39, please feel free to review the code. Thanks!

aravind14feb, 2016-11-21:
@haoxi911 
This is the change I did to resolve Short name issue.
In ShortName.java, 
`	@Override
	public int hashCode() {
		return getString().hashCode();
	}
`
Feel free to give me a feedback.

@magnusja  I was able to reproduce this junk file issue in Lenovo 5.0.1 tablet, Nexus Marshmallow

aravind14feb, 2016-12-05:
@magnusja  Any sort of leads to this issue of junk files,  will be really helpful.

haoxi911, 2016-12-07:
@aravind14feb I observed junk files again on a Galaxy S5 running Android 5.0.

1) I reproduced this issue by copying two 200 mb files onto an empty USB drive (through libaums), and on Windows, locate the two files, and duplicate them into 4 and 8 files, then the error dialog appeared. Do you think if you can give a try using the same steps?

2) Did you also observe the logs as below, in my side, these logs appeared hundreds of times:

```
W/Partition: device offset not a multiple of block size
W/ScsiBlockDevice: we have to round up size to next block sector
```

Thanks!
Kevin

[image]

magnusja, 2016-12-08:
The warnings indicate that you are not reading or writing at the position/index of a multiple of the block size of the file system. This usually should not be a problem, but it is a performance decrease. We first have to read some bytes, then merge these with the new data. Maybe there is an error when calculating this and this is causing the junk files.
https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/driver/scsi/ScsiBlockDevice.java#L232

magnusja, 2016-12-09:
One question: do you have this problem too when not using the stream classes?

magnusja, 2016-12-09:
Hey guys a little update:
Earlier this day I thought I found the bug: I created a new branch for that: https://github.com/magnusja/libaums/tree/bugfix/junk-files-issue-37
What basically happens is if you want to write a file from offset 0 to, for instance, 132, then this needs to be aligned with the block size of the device usually 512. What happens is that I create a new buffer with size 512 copy bytes 0 to 132 into that and write it to the device. As a result existing data from 132 to 512 gets overwritten with 0s from the buffer. This is bad, but I still don't know if this is a real problem, because usually files are allocated in 4096 chunks in FAT32. This means if you don't use the whole chunk any data after the last byte is ignored anyways. So I am not sure if it is possible that this leads to errors. Feel free to check out this branch and play around with it and see if the error persists.

I also created a branch where I copy files to USB with UsbFileOutputStream (sample app). Maybe you want to try if you are also getting these errors there. 
https://github.com/magnusja/libaums/tree/feature/copy-file-to-usb

aravind14feb, 2016-12-16:
@magnusja Sorry about the delayed response.

I have pulled the code from your branch for junk files issue. I believe the fix that you made might solve the corrupt file problem that @haoxi911  has reported in his last message.

Do you think the changes you have made can also solve the Junk files being generated?? (screenshot attached on Oct27)

We faced this junk files generation not just during the file write scenarios.. In certain cases, just reading the file structure from USB also resulted in the junk files. A separte file with junk text as names and a size of 0kb is generated.

Meanwhile we will try importing the changes u have made in the Junk files branch and see if we are able to reproduce the junk file problem.

magnusja, 2016-12-16:
@aravind14feb While reading too, that is interesting. That should not happen. There must be write commands somehow.

Just try develop and that branch and see if it persists. I am still unable to reproduce that and I played around with the library a lot lately.

aravind14feb, 2017-01-05:
@magnusja Sorry for the delayed update

We were able to reproduce junk files even with the code from the branch that you have created for it. The test case we followed was this

1. Copy a set of files from sdcard to USB 
(e.g /sdcard/Libaums/data1 , /sdcard/Libaums/data2)

2. After copying, connect it to a windows machine

3. Manually copy a set of files ( 9 or 10 files) to Libaums/data1 and Libaums/data2 directory , the copy might not be successful and will result in generation of junk files data.

If you could provide your email id, I can share you the source code and the apk of our sample app.

magnusja, 2017-01-05:
Is step 3 on the Windows machine?

aravind14feb, 2017-01-06:
@magnusja  Yes.

magnusja, 2017-01-09:
So this basically means that Windows creates these junk files and not my library?

aravind14feb, 2017-01-12:
@magnusja  I dont think so. Windows creates these junk files only after we make a file write using the libaums library.

magnusja, 2017-01-12:
Hmm okay. Which Windows version?

aravind14feb, 2017-01-12:
@magnusja Windows 7 and Windows 10

magnusja, 2017-01-19:
All right, I will try to spin up a VM with Windows, let's see.

magnusja, 2017-02-28:
Hey guys, I just released 0.5.1

Can you test once again for me, please. The read and write command length was too long. Maybe this caused issues with some flash drives.

46472761, 2017-04-06:
JAMES LACONIA DAVIS,TRIPLE ELITE,46472761

magnusja, 2017-05-18:
Hey, guys, I think I fixed this, can you test the current develop branch? Thanks!

closed 2017-05-18 by magnusja