magnusja/libaums #38: Get the file path

rameshvoltella, 2016-10-30:
How to get the file path of  file is usb using libaums

magnusja, 2016-11-01:
Hey,

seems that this is currently not implemented. But you could traverse thourght getParent() in combination with getName() to get the path.

Feel free to submit a PR.

rameshvoltella, 2016-11-02:
hi when use get parent it returns

code:
UsbFile entry=some value;
entry.getParent();

LOG BELOW
11-02 09:55:00.476 15782-15782/ D/LOG: com.github.mjdev.libaums.fs.fat32.FatDirectory@6c90955

and if i call **entry.getParent().getName()**

code:
UsbFile entry=some value;
entry.getParent().getName());

App crashes with null point Log below

LOG BELOW
     java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String com.github.mjdev.libaums.fs.fat32.FatLfnDirectoryEntry.getName()' on a null object reference

how can i fix this and get a readable path so that we can open the files using android intents services

magnusja, 2016-11-02:
You have to stop when getParent() == null, because that's the root directory. Use a while loop like this (untested):

```
while((String current = entry.getParent() != null)
   path = current + "/" + path
```

rameshvoltella, 2016-11-03:
but the path return is not a as a other filestsyem its return like
**com.github.mjdev.libaums.fs.fat32.FatDirectory@6c90955**

but am getting the file name no issues with that, so if need to open a file for example a video file using intent service  its not working so like the path is like **com.github.mjdev.libaums.fs.fat32.FatDirectory@6c90955/Test.mp4** if i made change as per the solution described but when i call intent to play this path file it fails.

Copying data to the phone and then play as per the example is not a good solution i think coz if my mobile storage has only 100 mb left the file i need to play is 1 gb in that case it will be a total failure, can you suggest a solution for this isuue?

magnusja, 2016-11-10:
Okay, maybe my code should be changed to this:

```
while((UsbFile current = entry.getParent() != null)
   path = current.getName() + "/" + path
```

You cannot easily pass these paths to an Intent because it does not know how to reach the USB drive. You can have a look at Android storage providers and then use the implementation in storageprovider/ to achieve what you want.

Some Android systems support mounting USB devices natively (e.g. Samsung). In this case, you do not need this library at all.

closed 2016-11-22 by magnusja

StormLion, 2017-07-18:
I have one more question, when connect a usb device to Android, we can get the path starts like this : /storage/usb(id=69B9-0F24/DCIM, so we then can use FileInputStream and FileOutputStream to read/write the file in the usb device, but now by entry.getParent().getParent()..., we can only get the root path in the specific device, but can't get it's absolute path on the Android. So if it's possible to get the path like /storage/usb(id=69B9-0F24/... from a UsbFile object.

magnusja, 2017-07-18:
I already explained that couple of times now (https://github.com/magnusja/libaums/issues/80, https://github.com/magnusja/libaums/issues/90, https://github.com/magnusja/libaums/issues/38, https://github.com/magnusja/libaums/issues/70). THIS IS NOT POSSIBLE! 

Your Android device happens to support accessing mass storage devices natively which is completely unrelated to this library. Please use UsbFileInput/OutputStream

rahultechwin, 2018-01-17:
It is also not work for me , I am only getting the name of image but don't get the complete path of image, 
can you please tell me how i'll get the complete path of images.

magnusja, 2018-01-17:
You can either use the current develop branch which features a method `getAbsolutePath` or you can use the code snipped I posted above.