magnusja/libaums #40: Folder names are displayed in all caps.

aravind14feb, 2016-11-29:
If all the characters in a folder name are lower case letters, then the libaums library is displaying them in ALLCAPS.

[image]

[image]

on Running with Storage Access Framework

[image]

Problem is consistent and easily reproducible.

magnusja, 2016-12-15:
Hey, so I tried that.
Unfortunately when I am creating a folder with all caps or with all lower case it still only creates a usual directory entry for both cases. As far as I know the FAT32 spec for the latter case there should be an lfn entry, but apparently, it is not created or there is a bug in my code somewhere.

So if you have any idea this is highly appreciated.

magnusja, 2016-12-15:
Found it: https://en.wikipedia.org/wiki/8.3_filename#Compatibility

closed 2016-12-15 by magnusja