magnusja/libaums #41: How to wite files to usb from internal storage

rameshvoltella, 2016-12-08:
Hi how can we write files from internal storage to usb.i have copy files from usb to internal storage but dont find how to copy files from internal to the usb storage. can you please help

rameshvoltella, 2016-12-08:
i try to apply the file creation logic to write from internal storage to usb but i failed here is what i use

` File fileFromInternalStorage= new File("/storage/emulated/0/Android/dm.txt");
      
UsbFile root=THIS IS ROOT OF THE CURRENT USB
      InputStream in = new FileInputStream(fileFromInternalStorage);
            byte[] bytes = IOUtils.toByteArray(in);
            root.write(0, ByteBuffer.wrap(bytes));
            root.close();`

But App is crashing

> Caused by: java.lang.UnsupportedOperationException: This is a directory!
>                                                                                 at com.github.mjdev.libaums.fs.fat32.FatDirectory.write(FatDirectory.java:560)

IOUtils.toByteArray(in); is from http://commons.apache.org/proper/commons-io/download_io.cgi

can you tell me what am doing wrong. can you help with a solution
Thank you

rameshvoltella, 2016-12-08:
New change

       File fileSource= new File("/storage/emulated/0/example2.mp4");

            
           Log.d("TAG",fileSource.getAbsolutePath());

            InputStream in = new FileInputStream(fileSource);

            byte[] buf = new byte[1024];
            int len;
            UsbFile file = rootBase.createFile("example2.mp4");//rootBase is current usbFile
          
            while ((len = ina.read(buf)) > 0) {
                Log.d("TAG","INSIDE");
                file.write(0, ByteBuffer.wrap(buf));
            }
            Log.d("TAG","OUTSIDE");
            file.close();

When i do this error has gone and file creates but its size is 1 kb and lot of corrupted files appear in directory.

Am i close to the solution(i know some thing is wrong :) ) or doing wrong. Please give a look at the code and please suggest a solution

magnusja, 2016-12-08:
Hey @rameshvoltella,

good that you came that far on your own!
Regarding the warnings: this is not a problem, it is just a performance decrease if you don't read and write multiple of the block size.

Regarding your file.write: you are always writing at position 0, which means the data from the last iteration get constantly overwritten. Have a look how I do it here: https://github.com/magnusja/libaums/blob/develop/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.java#L655

You may also want to have a look at: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/UsbFileOutputStream.java
This is taking care of the increasing index automatically. I think there are also easy ways to copy one stream into another with this OutputStream implementation.

Regarding the corrupted files: There is already an issue for that. #37

closed 2016-12-08 by magnusja

rameshvoltella, 2016-12-08:
soory about the code i forget to update the code actually
i tried
  while ((len = ina.read(buf)) > 0) {
            Log.d("TAG","INSIDE");
            file.write(len, ByteBuffer.wrap(buf));
        }

but still its gives same result

magnusja, 2016-12-08:
Yout might want to try read and write 4096 byte chunks.
If you see the block size warning it is likely that you are affected by issue #37. I am working on that. For now you can avoid that by reading and writing always 4096 byte chunks.

rameshvoltella, 2016-12-08:
so how to use UsbFileOutputStream

like this

       File fileSource= new File("/storage/emulated/0/example2.mp4");

        
       Log.d("TAG",fileSource.getAbsolutePath());

        InputStream in = new FileInputStream(fileSource);
       ByteBuffer buffer = ByteBuffer.allocate(4096);
            int len;
            UsbFile file = rootTemp.createFile("example2.mp4");
            UsbFileOutputStream mOutPut=new UsbFileOutputStream(file);
}
            while ((len = in.read(buffer2.array())) > 0) {
                Log.d("TAG","INSIDE");
                mOutPut.write(len);
            }
            Log.d("TAG","OUTSIDE");
            ina.close();
            mOutPut.close();

magnusja, 2016-12-08:
Now you are writing the int variable `len` to the file but you want to write the buffer of length len.

So try something like mOutPut.write(buffer.array());

You may also want to use just this: https://commons.apache.org/proper/commons-io/javadocs/api-2.4/org/apache/commons/io/IOUtils.html#copy(java.io.InputStream,%20java.io.OutputStream)

rameshvoltella, 2016-12-08:
Ya man its fixed Whooo

File fileSource= new File("/storage/emulated/0/example2.mp4");

    
   Log.d("TAG",fileSource.getAbsolutePath());

    InputStream in = new FileInputStream(fileSource);
   ByteBuffer buffer = ByteBuffer.allocate(4096);
        int len;
        UsbFile file = rootTemp.createFile("example2.mp4");
        UsbFileOutputStream mOutPut=new UsbFileOutputStream(file);
 }
        while ((len = in.read(buffer.array())) > 0) {
            Log.d("TAG","INSIDE");
            mOutPut.write(buffer.array());//This the key Point
        }
        Log.d("TAG","OUTSIDE");
        ina.close();
        mOutPut.close();