magnusja/libaums #63: no device

duwurensheng010, 2017-02-22:
Why i can not get my device?

duwurensheng010, 2017-02-22:
I connect with Digital camera。

magnusja, 2017-02-22:
Is there no partition or no device at all?

Can you maybe install the UsbDevice Info App and post some screenshots with information? Especially interesting are device class and the endpoints.

https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator&hl=en

duwurensheng010, 2017-02-23:
https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator&hl=en
I can get device info with this app，Device path is not the file path？

magnusja, 2017-02-23:
So what is the device info? Can you post some screenshots, please?
Device path something low level, which is not of interest.

duwurensheng010, 2017-02-23:
[image]

magnusja, 2017-02-23:
So the device class "Still Image Device (0x6)" is not supported in libaums. libaums only supports USB Mass Storage devices (https://en.wikipedia.org/wiki/USB_mass_storage_device_class).
If you cannot find a USB interface with that class, your camera does not support simulating such a device. What you can try then is using an SD card reader instead.

duwurensheng010, 2017-02-23:
Oh, I know, thank you very much

magnusja, 2017-02-23:
You are welcome!

closed 2017-02-23 by magnusja