magnusja/libaums #70: Uri to file path or usbFile to File

Qamar4P, 2017-03-14:
Can't get path for usbFile
I try to extract thumbnail for video using 
`InputStream is = new UsbFileInputStream(usbFile);` not posible. 
Further problem is to play that video clip from USB in VideoView without copying.

magnusja, 2017-03-14:
File path #38

How exactly are you trying to get the thumbnail?

To avoid copying you can either use the http or the storageprovider module.

Qamar4P, 2017-03-15:
No success yet but hope to get thumbnail using following. https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/FrameGrabber.java

Qamar4P, 2017-03-15:
Couldn't open file on client side; trying server side: java.io.IOException: setDataSourceFD failed.: status=0x80000000
03-15 14:39:42.183 7616-7616/com.citrusbits.leefaccess E/MediaPlayer: Unable to create media player
03-15 14:39:42.185 7616-7616/com.citrusbits.leefaccess W/VideoView: Unable to open content: content://com.citrusbits.leefaccess.usbstorage.documents/document/usb1002%3A%2FGOPR0462.MP4
                                                                    java.io.IOException: setDataSource failed.: status=0x80000000

magnusja, 2017-03-15:
Are you using the storage provider framework?

It seems that your VideoView does not support these kind of paths. Maybe try the http module instead.

Qamar4P, 2017-03-16:
@magnusja 
1. Yes, I am using [`UsbStorageProvider`](https://github.com/1hakr/AnExplorer/blob/master/app/src/main/java/dev/dworks/apps/anexplorer/provider/UsbStorageProvider.java)` extends DocumentsProvider`

2. [CommonsWare](www.github.com/commonsguy) mention that embedded web server has security ramifications [ref](http://stackoverflow.com/a/42784870/5710872). And further he said he will work with you on this library.

3. I also try wseemann/FFmpegMediaMetadataRetriever.

```
FFmpegMediaMetadataRetriever mmr = new FFmpegMediaMetadataRetriever();
                        mmr.setDataSource(LeefApp.getInstance(),documentUri);
//                        mmr.setDataSource(fd);
                        bm = mmr.getFrameAtTime(2000000);
```

thank you. It's working now.

4. How to use this uri in video player and seek back and forth without any issue without copying it's content? Meanwhile I am working on it. You suggestion will help me.

Qamar4P, 2017-03-17:
@magnusja I am available for research and help to solve URI and file path issue.

magnusja, 2017-03-17:
Hey there,

1. There is no need to use this class from the AnExplorer project. This officially feature is supported. You only have to install the module libaums-storageprovider. See the documentation for more information on that.

2. I don't know what exactly he means. If you use the NanoHTTP library, you can set the address to 127.0.0.1, so only apps on the device itself can reach the server and no one else in the network. Other than that I cannot see any issues, maybe you can ask him instead?

4. Are you trying to play the video in your app or are you opening a 3rd party app via an intent? For the latter, it might be worth checking out the libaums-http module. Usually, 3rd party media players all have support for playing http URLs. If you are trying to embed the player in your app, you either have to find something that supports URIs from ContentProviders or you also try the http module.

Let me know how your tests go, thanks!

Qamar4P, 2017-03-17:
Thanks, 1/2 I was using. Fine.
3/4 using embedded player. I try Android media player. It was not working with uri. I will let you know if any player works with your lib.

Qamar4P, 2017-03-27:
@magnusja ExoPlayer is working but buffering more than ES Explorer built-in player.

46472761, 2017-04-06:
JAMES LACONIA DAVIS,TRIPLE ELITE,46472761

fly7632785, 2017-06-06:
@magnusja 
hi,what a awesome framework.
but I need filePath ,too. Coz, It's forced by my Api to upload a File ,not a inputStream.  
So, I need filePath.
Is there any way to solve it? please.

magnusja, 2017-06-06:
Hey there,

nope unfortunately not. Because libaums cannot be integrated into the Android/Linux file handling :(

fly7632785, 2017-06-13:
Hey Magnusja
@magnusja 
May I ask you some questions?
There is no Api in android official developer web to operate the file system directly,isn't it?
Thanks to your open framework , I can read files from file system.
But I wanna know is there any other framework or official Api I didn't discover can operate files via filePath? 
After I checked the official Api , I figure it out that it transfers via bulkTransfer like socket, not via file system like your framework. right?
Why doesn't the official Api support it?

magnusja, 2017-06-13:
Hey there,

no there is no official Android API for accessing USB drives, but some devices mount USB drives and you can access them via the standard Java File API.
There is no API which exposes a file path, which is known to the rest of the OS, besides if the device mounts the drive on Linux kernel level.
The Android USB Host API offers very low level functionality in regards to USB (bulk transfer, ...). libaums implements several protocol layers to be able to expose a file system like API.

fly7632785, 2017-06-13:
@magnusja   
yes, what a great framework!
Thank you

fly7632785, 2017-06-22:
Hi,magnusja @magnusja 
You are one of great developers I know, I want to learn some from you. 
Maybe I ask you a few questions about usb equipment?
My Requirement:
I got a usb equipment(https://gyazo.com/d5c5288d0535ae83e9fb2cdb80adc55d), and I need read its files and upload files by android phone connecting to the usb equipment
But it can't  getPartitions, it's empty.
The usb equipment is Fat32.
As I know, the equipment maybe based on a linux system. So can I operate its file via Java File API? (besePath '/sys/bus/usb/devices/')
I know very little about linux usb equipment
hope get your help
thank you

magnusja, 2017-06-22:
If you are able to get a `UsbMassStorageDevice` via `getMassStorageDevices`, then your equipment should be fine. I am not sure what exactly is shown in that picture, but it says USB OTG and that is exactly what you need.
Are you sure that it is FAT32? 
Are you suing Linux or Windows? Maybe you can show me some details of the drive you are using.

fly7632785, 2017-06-22:
@magnusja  
The customer tells it's FAT32, but partitions are empty, that's weird. 
The usb equipment is not like a Usb storage
It can take photo and save it, if it connects to phone, then read files and upload.
Is there any way to judge it is FAT32 or not ?

magnusja, 2017-06-22:
Do you have a link to the product?
Hard to tell, can you use it on your computer?

fly7632785, 2017-06-22:
Sorry, I don't know now
I was confirming to the customer
The customer shows me some screenshots of APP( USB Device Info) connected to the usb equipment
[image]
 
[image]

[image]
[image]
[image]

what can they tell us?
It can read the device info
Thank you for your patience again

magnusja, 2017-06-22:
Yeah, that only says it is a mass storage device. But it cannot say what kind of file system.
Best would be to have a website about the product to get more information.

fly7632785, 2017-06-22:
I will confirm it tomorrow 
If it's really not FAT32, how could I do?
0_0

magnusja, 2017-06-22:
Connect it to your computer and see what it says about the file system

fly7632785, 2017-06-23:
hi, Magnusja.
Sorry for bothering you, too
I have a question.
When I use a Usb storage, it works good. We can get device info via android Api. But can't get devices via linux path "/sys/bus/usb/devices"  (in APP USB Device info)
However, the customer's usb equipment can get device info via android Api and  linux path
[image]
its device info in linux is like this:
[image]

I want to know why?
Is the usb equipment a linux usb or something else? 
Why does it have 3 devices under the path of "/sys/bus/usb/devices/", but normal Usb storage has nothing?

magnusja, 2017-06-23:
A USB device can have multiple USB interfaces. Imagine a USB hub, it is a USB device itself but represents the devices you connect to it.

fly7632785, 2017-06-23:
yes, you are right 
So the usb equipment has a storage, too. Like a normal  Usb storage, we can access its fileSystem.
And its fileSystem type is FAT32, what can cause that it gets empty partition?

magnusja, 2017-06-23:
Usually, that is the case if the device is not formatted with an MBR and a file system supported by libaums.

So make 100% sure that this is the case. How do you know it is FAT32?

fly7632785, 2017-06-25:
Yes, it is FAT32. When I put it on the computer, it shows.
[image]
[image]
Sorry, I just don't know how to do.

magnusja, 2017-06-27:
Can you maybe create an image for me and send it to me via email?
´dd if=/dev/YOURDRIVE of=usb.img bs=10m count=1´
Run the following command to get the path to your drive on a mac:
´diskutil list´

fly7632785, 2017-06-28:
what you mean is to connect Usb equipment to computer, and execute 'diskutil list'?  
As to 'dd if=...', I don't know what my drive is.
Should I execute 'dd if=....', too?
Sorry, and I don't know your email.

By the way, the Usb equipment can be read by the APP(https://play.google.com/store/apps/details?id=com.homeysoft.nexususb.importer&hl=ja)
The App is very great, I don't think I can implement the same app like that.So I can just count on your help.
Thank you

fly7632785, 2017-06-28:
The command result:
bin-no-MBP:temp bingao$ dd if=/Volumes/NO\ NAME/ of=usb.img bs=10 count=1
1+0 records in
1+0 records out
10 bytes transferred in 0.000082 secs (121927 bytes/sec)

B00???bin-no-MBP:temp bingao$ diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Macintosh HD            250.1 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Macintosh HD           +249.8 GB   disk1
                                 Logical Volume on disk0s2
                                 8ABB2546-043F-4E17-A20A-8EECAB36BE00
                                 Unlocked Encrypted

/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                            NO NAME                *7.9 GB     disk2

fly7632785, 2017-06-28:
The NO NAME is the USB equipment. 
Is there any valuable information?

fly7632785, 2017-06-28:
Hi, magnusja
There's a more info about the USB equipment.
It has a micro SD card  in it. Data is saved in it.
Is this reason that we can't read from it?

fly7632785, 2017-06-28:
Because I don't have your e-mail. Here is the usb.img file(http://jafir-my-love.oss-cn-shanghai.aliyuncs.com/usb.img)

magnusja, 2017-06-28:
Yes, I think SD card readers do not work at the moment for some reason. Maybe I find some time in the next couple of days to do some tests.

fly7632785, 2017-06-29:
OK, thank you!
You mean it originally supports it?
Is there anything I can do for you?

fly7632785, 2017-06-29:
http://jafir-my-love.oss-cn-shanghai.aliyuncs.com/log.txt
this is a log
It doesn't seem it has any abnormal log

fly7632785, 2017-06-29:
And  I  get some information.
[image]
the partitionType is always 0  four times 
so 'result.partitions.add(entry) ' didn't execute.

fly7632785, 2017-06-29:
More information is that the USB equipment may doesn't have a MBR support 
[image]

closed 2017-07-03 by magnusja

honeyvinay95, 2020-02-20:
I have made an application using "libaums" to get files from external device but  could not able to get the Path.

package com.example.usbread;

import androidx.appcompat.app.ActionBar;
import androidx.appcompat.app.AlertDialog;
import androidx.appcompat.app.AppCompatActivity;

import android.app.PendingIntent;
import android.content.BroadcastReceiver;
import android.content.ComponentName;
import android.content.Context;
import android.content.DialogInterface;
import android.content.Intent;
import android.content.IntentFilter;
import android.content.ServiceConnection;
import android.hardware.usb.UsbDevice;
import android.hardware.usb.UsbManager;
import android.os.Bundle;
import android.os.IBinder;
import android.util.Log;
import android.view.View;
import android.widget.AdapterView;
import android.widget.Button;
import android.widget.ListView;
import android.widget.Spinner;
import android.widget.Toast;

import com.github.mjdev.libaums.UsbMassStorageDevice;
import com.github.mjdev.libaums.fs.FileSystem;
import com.github.mjdev.libaums.fs.UsbFile;
import com.github.mjdev.libaums.server.http.UsbFileHttpServerService;
import com.koushikdutta.async.future.Cancellable;

import java.io.IOException;

public class MainActivity extends AppCompatActivity implements AdapterView.OnItemClickListener {
    private static final String ACTION_USB_PERMISSION = "com.example.usbread.USB_PERMISSION";
    private static final String TAG = "MemoryListing";
    private int currentDevice = -1;
    private FileSystem currentFs;
    private UsbFileHttpServerService serverService;
    UsbMassStorageDevice[] massStorageDevices;
    UsbFileListAdapter adapter;
    Spinner spinner;
    ListView list;
    Button button;

    private final BroadcastReceiver usbReceiver = new BroadcastReceiver() {      //Broadcast receiver is an Android component which allows you to send or receive Android system or application events

        @Override
        public void onReceive(Context context, Intent intent) {
    String action = intent.getAction();
			if (ACTION_USB_PERMISSION.equals(action)) {

        UsbDevice device = (UsbDevice) intent.getParcelableExtra(UsbManager.EXTRA_DEVICE);
        if (intent.getBooleanExtra(UsbManager.EXTRA_PERMISSION_GRANTED, false)) {

            if (device != null) {
                setupDevice();
            }
        }

    } else if (UsbManager.ACTION_USB_DEVICE_ATTACHED.equals(action)) {
        UsbDevice device = (UsbDevice) intent.getParcelableExtra(UsbManager.EXTRA_DEVICE);

        Log.d(TAG, "USB device attached");

        // determine if connected device is a mass storage devuce
        if (device != null) {
[... 177 lines cut ...]
            Log.d(TAG, "Occupied Space: " + currentFs.getOccupiedSpace());
            Log.d(TAG, "Free Space: " + currentFs.getFreeSpace());
            Log.d(TAG, "Chunk size: " + currentFs.getChunkSize());
            UsbFile root = currentFs.getRootDirectory();

            Toast.makeText(this,currentFs.toString(),Toast.LENGTH_LONG).show();
            Log.d("testing",root.toString());

            ActionBar actionBar = getSupportActionBar();
            actionBar.setTitle(currentFs.getVolumeLabel());

            list.setAdapter(adapter = new UsbFileListAdapter(this, root));
        } catch (IOException e) {
            Log.e(TAG, "error setting up device", e);
        }

    }

    @Override
    public void onItemClick(AdapterView<?> adapterView, View view, int i, long l) {
        setupDevice();
    }
}

[image]

magnusja, 2020-02-20:
See `UsbFile.getAbsolutePath()`

honeyvinay95, 2020-02-21:
@magnusja  I have searched getAbsolutePath in UsbFile interface but you did not define any this method in your interface 
[image]

magnusja, 2020-02-21:
You are using a pretty old version.
https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/UsbFile.kt#L63

honeyvinay95, 2020-02-21:
Hi @magnusja I have searched everywhere how to get the Path of an External Device could not find any solution :(  The issue is when I connect USB to android device I got a list of files which is inside my USB stick and want to show the Path of an Each file which is inside my USB stick like this storage/USB A/filename but I could not able to get the path I hope you will resolve my problem my last hope is you......

magnusja, 2020-02-21:
Not sure what you need. You get the absolute path and then you can prepend the USB devices name or waht not if you want and show that to make it clear it is on a USB device.

honeyvinay95, 2020-02-22:
> You are using a pretty old version.
> https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/UsbFile.kt#L63

[image]

I'm using Java i dont have any idea of Kotlin

honeyvinay95, 2020-02-22:
> Not sure what you need. You get the absolute path and then you can prepend the USB devices name or waht not if you want and show that to make it clear it is on a USB device.

I want to get absolute path of an External Device or USB, When I connect the USB to android Device it will show me the Path like the below image:
[image]

[image]

honeyvinay95, 2020-02-24:
@magnusja 

[image]

magnusja, 2020-02-24:
If this is only for displaying purposes I would go with the solution I posted earlier. 

There is no Android accessible path when your are using libaums. It is just technically not feasible.

honeyvinay95, 2020-02-24:
@magnusja By mistake I wrote the wrong  sentence pardon my English but I need to get External Storage Path. In my code I have tried so many ways to get the External Storage Path getAbsolutePath(),  getRootDirectory() and Environment.getExternalStorageDirectory() but could not able to get the External Storage Path

magnusja, 2020-02-24:
No you are not.

You can get the absolute path within a USB directory described in #38.

The general Android USB path is out of scope for this library. There are devices which support that and some other do not. The question is why would you need the path? If you only want to display it, why does it need to be the Android USB path, why not something else where the user gets the idea that the file is in some folder on the USB drive.

honeyvinay95, 2020-02-27:
Thank you @magnusja  finally got the path :)

Shantanu-tec, 2022-08-06:
Hi,
I am using this library to get the contents stored in the USB. It is working fine.
But when i need to decrypt the content or play the content. I need an absolute path of the USB in android file system.

Can you guys please help me.

surjansinghdonkeli, 2023-03-18:
[image]

Sir I have found pendrive path but folder is not able to be found please help me thanks sir

surjansinghdonkeli, 2023-03-18:
using this code

 public void listDirectory(String dirPath, int level) {
        File dir = new File(dirPath);
        

        Toast.makeText(getApplicationContext(),  String.valueOf(dir), Toast.LENGTH_SHORT).show();
        
        File[] firstLevelFiles = dir.listFiles();
        if (firstLevelFiles != null && firstLevelFiles.length > 0) {
            for (File aFile : firstLevelFiles) {
                for (int i = 0; i < level; i++) {
                    System.out.print("\t");
                }
                if (aFile.isDirectory()) {
                    System.out.println("[" + aFile.getName() + "]");
                    
                    
                    listDirectory(aFile.getAbsolutePath(), level + 1);
                } else {
                    System.out.println(aFile.getName());
                    
                    
                    
                    
                }
            }
        }
}

magnusja, 2023-03-20:
You are not even using this library?! Why would you ask that here. Please ask that on stackoverflow or sth ...

surjansinghdonkeli, 2023-03-24:
Sir Your Library Display This

[image]

surjansinghdonkeli, 2023-03-24:
Dear Sir,
             My requirement Data Copy Pen drive to Mobile Device. How Can Possible. Please Suggest Me as Soon as Possible.

Thanks & Regards,
Surjan Singh
9958715332

magnusja, 2023-03-24:
That is an old link https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/me/jahnen/libaums/core/fs/UsbFile.kt

Please consult the documentation it is all explained there. In addition, try the example application from this repository.