magnusja/libaums #71: can libaums support multiple devices connect simultaneously

StormLion, 2017-03-15:
Hi, now I'm need to handle multiple devices at the same time, but I don't known if this library support it.

magnusja, 2017-03-15:
Hey there,

I have never really tried it, but it should work. Just try it and tell me if you have any problems.

StormLion, 2017-03-16:
Hi, I've tried it many times, I used a usb hub to connect to several devices at the same time. I can got all the dirs of all the devices. 
[image]
As the picture shows, I use two listview, one use to show all the connected devices, the other one use to show the content of the specific deivce, when I clicked a specific device, I will refresh the content listview,
this worked fine, but when I clicked a specific folder in the content listview, I can't got the subdirectory of it and I got an error log as blow:
03-14 00:46:28.573 21986-21986/star.usbdevicescontroller W/System.err: java.io.IOException: Could not write to device, result == -1
03-14 00:46:28.573 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkOutTransfer(JellyBeanMr2Communication.java:42)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:143)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:209)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.partition.Partition.read(Partition.java:142)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:134)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:64)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.init(FatDirectory.java:156)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.listFiles(FatDirectory.java:485)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at star.usbdevicescontroller.usbmode.UsbFileListAdapter.refresh(UsbFileListAdapter.java:80)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at star.usbdevicescontroller.usbmode.UsbFileListAdapter.<init>(UsbFileListAdapter.java:60)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at star.usbdevicescontroller.usbmode.UsbActivity$2.onItemClick(UsbActivity.java:255)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.widget.AdapterView.performItemClick(AdapterView.java:310)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.widget.AbsListView.performItemClick(AbsListView.java:1145)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.widget.AbsListView$PerformClick.run(AbsListView.java:3066)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.widget.AbsListView$3.run(AbsListView.java:3903)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.os.Looper.loop(Looper.java:148)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5417)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
03-14 00:46:28.574 21986-21986/star.usbdevicescontroller W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:742)
03-14 00:46:28.575 21986-21986/star.usbdevicescontroller W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:632)
03-14 00:46:34.560 21986-21986/star.usbdevicescontroller E/UsbMassStorageDevice: could not release interface!
03-14 00:46:34.560 21986-21986/star.usbdevicescontroller E/UsbMassStorageDevice: could not release interface!
(I have also tried to solve it, but I can't got it work. I need your help, thanks advance.
I come from China, so my English is not very well, hope you can understand me :) )

StormLion, 2017-03-16:
And here are some relative code snippets:

    private void discoverDevice() {
        UsbManager usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(this);

        if (devices.length == 0) {
            ActionBar actionBar = getSupportActionBar();
            if(actionBar != null) {
                actionBar.setTitle(R.string.usb_no_devices);
            }
            listView.setAdapter(null);
            return;
        }

        // the type of storageDevices is List<UsbMassStorageDevice>
        // add all founded usb device to the recycler view
        for(int i=0; i<devices.length; i++) {
            storageDevices.add(devices[i]);
        }
        UsbDevice usbDevice = getIntent().getParcelableExtra(UsbManager.EXTRA_DEVICE);

        if (usbDevice != null && usbManager.hasPermission(usbDevice)) {
            setupDevice();
        } else {
            for(int i=0; i< storageDevices.size(); i++) {
                PendingIntent permissionIntent = PendingIntent.getBroadcast(this, 0, new Intent(ACTION_USB_PERMISSION), 0);
                usbManager.requestPermission(storageDevices.get(i).getUsbDevice(), permissionIntent);
            }
        }
    }

    private void setupDevice() {
        try {
            long sTime = System.currentTimeMillis();
            for(int i=0; i<storageDevices.size(); i++) {
                storageDevices.get(i).init();
            }

            long eTime = System.currentTimeMillis();
            textViewSpeed.setText("Init all devices, devices count = " + storageDevices.size() + ", use time " + (eTime - sTime) + "ms");

            UsbMassStorageDevice curUsbDevice = storageDevices.get(0);
            customAdapter.setStorageDevices(storageDevices);

            // we always use the first partition of the usbMassStorageDevice0
            FileSystem fileSystem = curUsbDevice.getPartitions().get(0).getFileSystem();

            UsbFile root = fileSystem.getRootDirectory();
            ActionBar actionBar = getSupportActionBar();
            if(actionBar != null) {
                actionBar.setTitle(fileSystem.getVolumeLabel());
            }

            listView.setAdapter(usbFileListAdapter = new UsbFileListAdapter(this, root));
        } catch (IOException e) {
            Log.e(TAG, "error setting up usbMassStorageDevice", e);
        }

    }

        onItemClickListener = new AdapterView.OnItemClickListener() {
            @Override
            public void onItemClick(AdapterView<?> parent, View view, int position, long rowId) {
                UsbFile entry = usbFileListAdapter.getItem(position);
                try {
                if (entry.isDirectory()) {

                    dirs.push(usbFileListAdapter.getCurrentDir());

                    // TODO multiple devices cause IOException at here
                    listView.setAdapter(usbFileListAdapter = new UsbFileListAdapter(UsbActivity.this, entry)); 
                    
                } else {

                    if (ContextCompat.checkSelfPermission(UsbActivity.this, Manifest.permission.WRITE_EXTERNAL_STORAGE) != PackageManager.PERMISSION_GRANTED) {

                        if (ActivityCompat.shouldShowRequestPermissionRationale(UsbActivity.this, Manifest.permission.WRITE_EXTERNAL_STORAGE)) {
                            Toast.makeText(UsbActivity.this, R.string.request_write_storage_perm, Toast.LENGTH_LONG).show();
                        } else {
                            ActivityCompat.requestPermissions(UsbActivity.this, new String[]{Manifest.permission.WRITE_EXTERNAL_STORAGE}, REQUEST_EXT_STORAGE_WRITE_PERM);
                        }
                        return;
                    }

                    CopyTaskParam copyTaskParam = new CopyTaskParam();
                    copyTaskParam.from = entry;

                    // ------ TODO set the file path at here
                    File file = new File(Environment.getExternalStorageDirectory().getAbsolutePath() + "/usbfileman/cache");

                    boolean ret = file.mkdirs();
                    int index = entry.getName().lastIndexOf(".") > 0 ? entry.getName().lastIndexOf(".") : entry.getName().length();
                    String prefix = entry.getName().substring(0, index);
                    String ext = entry.getName().substring(index);
                    // prefix must be at least 3 characters
                    if(prefix.length() < 3) {
                        prefix += "pad";
                    }
                        copyTaskParam.to = File.createTempFile(prefix, ext, file);
                        new CopyTask().execute(copyTaskParam);
                }
                }
                catch (IOException e) {
                    Log.e(TAG, "error staring to copy!", e);
                }
            }
        };

StormLion, 2017-03-17:
Hi, I found that when connect multiple devices, the JellyBeanMr2Communication Ojbect refrence to the wrong UsbDeviceConnection Object, but I can't find the specific position of the bug.

magnusja, 2017-03-17:
Hey there. Are you sure that is the case? This line creates the connection: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/UsbMassStorageDevice.java#L213

So I assume you are using a USB Hub, so maybe there is a shared device connection involved, because of that. I never tried that to be honest. But the error you have is unfortunately not very specific in what exactly failed.

What you maybe can do is change from the synchronous data transfer to the USB request mode like so:
```UsbCommunicationFactory.setUnderlyingUsbCommunication(UnderlyingUsbCommunication.USB_REQUEST_ASYNC);```

StormLion, 2017-03-20:
I can't closing any devices, because I need to copy data from all the devices at the same time.

StormLion, 2017-03-20:
And I've tried UsbCommunicationFactory.setUnderlyingUsbCommunication(UnderlyingUsbCommunication.USB_REQUEST_ASYNC);
but it didn't work

StormLion, 2017-03-20:
I tried to modify the motheds of the UsbFile interface like this
UsbFile[] listFiles(UsbDeviceConnection deviceConnection) throws IOException;
and modity the relative implements, then it worked, but I don't think this is a good way.

magnusja, 2017-03-20:
I don't think I understand, can you show some code?

StormLion, 2017-03-21:
To be honest, I don't really sure if I have a full understand of your idea, maybe I've wrong, so I need you guide. I've tried to make some debug in you project, and upload it as a temp project to github, soon I'll delete it. you can download it from
https://github.com/StormLion/tproject

magnusja, 2017-03-21:
Thanks, I downloaded it and will have a look as soon as I am home.

blood-mist, 2017-03-29:
If you get multiple device supporting done, please post a little  code snippet on how to achieve it. Thanks.

46472761, 2017-04-06:
JAMES LACONIA DAVIS,TRIPLE ELITE 46472761

magnusja, 2020-05-13:
Fixed with #242

closed 2020-05-13 by magnusja

Depau, 2020-05-13:
I'll add this one to the list of things I'll want to test on Qemu. This should also cover the case of devices with multiple USB ports without an external hub, for example on Android x86.