magnusja/libaums #73: install error

farzad-kafami, 2017-04-29:
hi i have problem with this library
I write a sample app to read usb mass storage when i attach usb sotrage and run
`FileSystem currentFs = device.getPartitions().get(0).getFileSystem();`
  device.getPartitions() is empty I debug that my partitionType is -1 why?(my file system is FAT32)
then I try to download and run your code but i faced this error:
Installation failed with message INSTALL_FAILED_CONFLICTING_PROVIDER.
It is possible that this issue is resolved by uninstalling an existing version of the apk if it is present, and then re-installing.
can help me

magnusja, 2017-04-29:
You can fix the INSTALL_FAILED_CONFLICTING_PROVIDER if you rename your provider in the AndroidManifest.xml or delete your app before trying to install the example.

Are you using the master version? If so, maybe try the latest stable version.

farzad-kafami, 2017-04-30:
I rename provider authorities in AndroidManifest.xml and installed app but still not working for me
I debug that an get this problem:
`
public class Fat32FileSystemCreator implements FileSystemCreator {

    @Override
    public FileSystem read(PartitionTableEntry entry, BlockDeviceDriver blockDevice) throws IOException {
        if(entry.getPartitionType() != PartitionTypes.FAT32) return null;

        return Fat32FileSystem.read(blockDevice);
    }
}`
entry.getPartitionType() is -1
my flash memory file system is FAT32
can help me

farzad-kafami, 2017-04-30:
I used of usb 3 when I usb 2 problem has been fixed
but still usb 3 has bug

magnusja, 2017-05-02:
Can you debug this line: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/partition/mbr/MasterBootRecord.java#L92

And then tell me what the value of byte partitionType is?

farzad-kafami, 2017-05-03:
my problem is by SiliconPower flash memory
I test a Sony and anything is ok
I debug, partionType is 114,101,121,13 (SiliconPower )

magnusja, 2017-05-10:
So the partitionType should only be one byte nothing more. Are you showing me an array?
Here (https://en.wikipedia.org/wiki/Partition_type) you can see that the value 13 is FAT32, all the other values don't indicate an FAT32 file system.

I just tested a Samsung USB 3 drive and it worked just fine.
Did you try reformatting the device?

closed 2017-07-03 by magnusja

diljil, 2018-02-08:
hi.... 
In case device.getPartitions().isEmpty(); is true what should i do?

magnusja, 2018-02-08:
Check if you are using MBR boot sector and FAT32 file system.

saeedakhtar, 2018-02-13:
Hi,
I have device.getPartitions().isEmpty(); true.
Here is the USB configuration;

_UsbConfiguration[mId=1,mName=null,mAttributes=224,mMaxPower=250,mInterfaces=[ UsbInterface[mId=0,mAlternateSetting=0,mName=Mass Storage,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[ UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0] UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=1]]]_

magnusja, 2018-02-13:
This does not help at all. I already mentioned you should make sure your disk is formatted with MBR and FAT32

saeedakhtar, 2018-02-14:
@magnusja I have MBR FAT32 USB device. But I cannot read the files. I've tried using ES Explorer and it is showing me files which I want to copy.

magnusja, 2018-02-14:
Well, then you have to debug why it is not recognizing your device as MBR with FAT32. The partitions list is only empty if it could not recognize any supported formats.