magnusja/libaums #75: When I finished, I had a lot of 0kb files

dllsndy, 2017-05-09:
I downloaded about more than and 100 small files, divided into several folders.
After the completion of the download more than a fixed number of 0kb files, and some will be deleted file name is invalid or too long. And I clone your project, found that there will be a problem, how to solve this problem?

dllsndy, 2017-05-09:
copy .not download

dllsndy, 2017-05-10:
/0003 has entries after end of directory. '/0003' is a directory.How should I solve it?

magnusja, 2017-05-10:
Can you show some code?

dllsndy, 2017-05-11:
this is write(); 
new Thread(new Runnable() {
                @Override
                public void run() { 
   UsbFile root = massDevice.getPartitions().get(0).getFileSystem().getRootDirectory();
                     for (File file : files) { if (file.isDirectory()) {
                                UsbFile orderDirectory = root.search(file.getName());
                                if (orderDirectory== null) {
                                  orderDirectory = root.createDirectory(file.getName());
                                }
FileManager.copyAllUsbFile(file.getPath(), orderDirectory, massDevice.getPartitions().get(0).getFileSystem());
 }).start();

and copyAllUsbFile(String,String,FS);

{
        File dir = new File(srcFileStr);
        if (!dir.exists()) {
            return false;
        }
        File[] files = dir.listFiles();
        byte[] buffer = new byte[currentFs.getChunkSize()];
        for (File file : files) {
            if (file.isDirectory()) {
                continue;
            } else if (file.isFile()) {
                Uri uri;
                if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.N) {
                    uri = FileProvider.getUriForFile(MyApp.getMyApplication(),
                            MyApp.getMyApplication().getApplicationContext().getPackageName() + ".fileprovider",
                            file);
                } else {
                    uri = Uri.fromFile(file);
                }
                InputStream input =MyApp.getMyApplication().getContentResolver().openInputStream(uri);
//                FileInputStream input = new FileInputStream(file);
                FatFile searchFile = (FatFile) root.search(file.getName());
                if (searchFile != null) {
                    L.e("delete" + searchFile.getName());
                    searchFile.delete();
                }
                 searchFile = (FatFile) root.createFile(file.getName());
                searchFile.setLength(file.length());
                L.i("msg createFile=", searchFile.getName());
              BufferedOutputStream out = UsbFileStreamFactory.createBufferedOutputStream(searchFile, currentFs);
 //                 UsbFileOutputStream out=new UsbFileOutputStream(searchFile);
                int len = 0;
                //TODO
                L.e("write" + searchFile.getName());
                while ((len = (input.read(buffer))) != -1) { 
                    out.write(buffer,0, len);
                }
                out.flush();
                input.close();
                out.close(); 
            }
        }
    }

dllsndy, 2017-05-12:
like this.
[image]

dllsndy, 2017-05-15:
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 155 continues with cluster 156 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 156 continues with cluster 157 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 157 continues with cluster 158 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 158 continues with cluster 159 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 159 continues with cluster 160 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 160 continues with cluster 161 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 161 continues with cluster 162 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 162 continues with cluster 163 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 163 continues with cluster 164 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 164 continues with cluster 165 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 165 continues with cluster 166 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 166 continues with cluster 167 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 167 continues with cluster 168 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 168 continues with cluster 169 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 169 continues with cluster 170 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 170 continues with cluster 171 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 171 continues with cluster 172 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 172 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 173 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 174 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 175 continues with cluster 176 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 176 continues with cluster 177 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 177 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 178 continues with cluster 179 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 179 continues with cluster 180 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 180 continues with cluster 181 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 181 continues with cluster 182 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 182 continues with cluster 183 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 183 continues with cluster 184 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 184 continues with cluster 185 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 185 continues with cluster 186 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 186 continues with cluster 187 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.445 432-442/? I/fsck_msdos: Cluster 187 continues with cluster 188 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.445 432-442/? I/fsck_msdos: FIXED
[... cut ...]
05-15 10:51:58.446 432-442/? I/fsck_msdos: Cluster 245 continues with cluster 246 in FAT 0, but is marked free in FAT 1
05-15 10:51:58.446 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.446 432-442/? I/fsck_msdos: Cluster 246 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.446 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.446 432-442/? I/fsck_msdos: Cluster 247 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.446 432-442/? I/fsck_msdos: FIXED
05-15 10:51:58.446 432-442/? I/fsck_msdos: Cluster 248 is marked as EOF in FAT 0, free in FAT 1
05-15 10:51:58.446 432-442/? I/fsck_msdos: FIXED

magnusja, 2017-05-16:
Hey there,

so I guess you have the same problem with junk files like #33 and #78. Have a look at #78, I commented something you can try for me.

magnusja, 2017-05-16:
Can you try current develop branch for me, please?

dllsndy, 2017-05-17:
ok, I will try it.

dllsndy, 2017-05-17:
This new example doesn't run into problems.
And there is no problem with my project importing ‘javafs’ and ‘libaums’.
I'm using API 25.

magnusja, 2017-05-17:
So problem solved,  I assume?

dllsndy, 2017-05-18:
Yes, thank you for helping me solve this problem.

closed 2017-05-18 by dllsndy