magnusja/libaums #77: createDirectory not working for Samsung devices

umair-brainx, 2017-05-13:
# The Following code doesn't create sampleFolder for Samsung devices

`dataFile = root.createDirectory("Data");
UsbFile sample=dataFile.createDirectory("sampleFolder");
`

umair-brainx, 2017-05-13:
I have tested the application on Sony Xperia Device it works fine but it doesn't work on Samsung devices. 
@magnusja Kindly let me know if I am doing something wrong.

magnusja, 2017-05-16:
Hey there,

issue #33 already discusses this problem with Samsung devices. I really don't know what is happening there. Can you tell which Samsung devices and which version of Android are affected?

umair-brainx, 2017-05-16:
I have tested it on Samsung Note 4(Android 6.0.1), Samsung J7 prime(Android 6.0.1) and Samsung Tab 2(probably android 6.0.1). Even USB light blink. I have also debugged the code it shows that FatDirectory has correct entries but there is no folder created on USB. And one important thing is when i create directory from root directory object it works but when I create directory from any sub folder of root it doesn't work.

umair-brainx, 2017-05-16:
@magnusja I have also read the all comments on issue #33. Thanks for sharing. Yes I think my issue is same. So now what do you think? How we can fix this issue?

magnusja, 2017-05-16:
I don't know. I guess I need to get one of these Samsung devices. Can you tell me the cheapest option on where this issue occurs?

umair-brainx, 2017-05-16:
I have tested only on these three devices. There are also cheaper devices than these where this issue may produce. I just searched craigslist you can find the same model i am using in 150$. i.e Samsung Note 4. This link might help you https://sfbay.craigslist.org/sby/mob/6133525277.html

lesniakania, 2017-06-30:
I have exactly the same problem. I'm testing on a tablet Samsung Galaxy Tab S2, model SM-T719 and a phone Galaxy S6, model SM-G920F. Both have Android 7.

magnusja, 2017-06-30:
Hey there,

sorry to hear that. I have very limited time at the moment and still did not buy a Samsung device. So I cannot give an estimate when this will be fixed. 
In the meanwhile, I would suggest trying to debug the issue yourself.

lesniakania, 2017-07-02:
It's not urgent enough to do this during my work hours and unfortunately I don't have much time after work for this at this moment. I just wanted to check status, no worries.

closed 2023-03-11 by magnusja