magnusja/libaums #78: Copy folder from internal storage to usb with progress

alashow, 2017-05-13:
Hello. Thanks for an amazing library!

I was trying to implement a function that recursively copies a folder from an internal storage to a USB drive.

Here is what I got now:
```java
CopyToUsbTaskParam params = new CopyToUsbTaskParam();
params.currentFs = currentFs;
params.from = Config.getDownloadsFolder(); /*new File("/sdcard/AppFolder")*/
params.toFolder = "files";

new CopyToUsbTask().execute(params);
```

```java
private static class CopyToUsbTaskParam {
        File from;
        String toFolder;
        FileSystem currentFs;
        boolean overwrite = true;
    }

    private class CopyToUsbTask extends AsyncTask<CopyToUsbTaskParam, Integer, Void> {

        private ProgressDialog dialog;
        private CopyToUsbTaskParam param;

        CopyToUsbTask() {
            dialog = new ProgressDialog(getMainActivity());
            dialog.setTitle("Copying files");
            dialog.setMessage("Copying files to the USB drive, this can take some time!");
            dialog.setIndeterminate(true);
            dialog.setProgressStyle(ProgressDialog.STYLE_HORIZONTAL);
        }

        @Override
        protected void onPreExecute() {
            if (dialog != null && ! dialog.isShowing()) {
                dialog.show();
            }
        }

        @Override
        protected Void doInBackground(CopyToUsbTaskParam... params) {
            param = params[0];

            try {
                UsbFile root = param.currentFs.getRootDirectory();

                // find destination folder or create it
                UsbFile to = param.currentFs.getRootDirectory().search(param.toFolder);
                if (to == null) {
                    to = root.createDirectory(param.toFolder);
                }
                copyDirectory(param.from, to);
            } catch (IOException e) {
                Logger.e(e, "Error copying folder: %s", e.getLocalizedMessage());
            }
            return null;
        }

        private void copyDirectory(File source, UsbFile destination) throws IOException {
            if (source.isDirectory()) {
                if (destination.isRoot()) {
                    destination = destination.createDirectory(source.getName());
                }

                String[] children = source.list();

                for(String child : children) {
                    if (param.overwrite) {
                        deleteIfExists(destination, child);
                    } else if (destination.search(child) != null) { // skip if already exists
                        continue;
                    }
                    File childSource = new File(source, child);
                    UsbFile childDestination = childSource.isDirectory() ? destination.createDirectory(child) : destination.createFile(child);
                    copyDirectory(childSource, childDestination);
                }
            } else {
                InputStream inputStream = new BufferedInputStream(new FileInputStream(source));
                OutputStream outputStream = UsbFileStreamFactory.createBufferedOutputStream(destination, param.currentFs);

                byte[] bytes = new byte[4096];
                int count;

                while ((count = inputStream.read(bytes)) != - 1) {
                    outputStream.write(bytes, 0, count);
                }

                outputStream.close();
                inputStream.close();
            }
        }

        private void deleteIfExists(UsbFile parent, String child) throws IOException {
            UsbFile file = parent.search(child);
            if (file != null) {
                file.delete();
            }
        }

        @Override
        protected void onPostExecute(Void result) {
            U.dismissDialog(dialog);
        }
    }

// util class
public class U {
public static void dismissDialog(DialogInterface alertDialog) {
        try {
            if (alertDialog != null) {
                if (alertDialog instanceof Dialog && ! ((Dialog) alertDialog).isShowing()) {
                    return;
                }
                alertDialog.dismiss();
            }
        } catch (final Exception e) {
            e.printStackTrace();
        }
    }
}
```

My problems/questions are:
1. Android Explorer shows junk files. Almost always. Doesn't show junk files in some folders.
But ES Explorer and Finder (osx) doesn't show anything.
Might be related to #37 
https://imgur.com/a/9PRon

2. How to properly add a total progress?

umair-brainx, 2017-05-16:
Hi I have the same issue. My code is working fine for Sony Xperia Device but facing similar issue on Samsung devices.

umair-brainx, 2017-05-16:
Let me know if you have figure out any solution?

alashow, 2017-05-16:
@umair-brainx what is your issue exactly? Junk files on system explorer?

magnusja, 2017-05-16:
So apparently more and more people are running into the junk files problem, unfortunately, I still don't know what the problem is.
Can you try this for me:
1. Check out current develop version, and preferably use my example app.
2. Please comment out this line: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/FileSystemFactory.java#L44
3. Make sure this line is included somewhere in the code: https://github.com/magnusja/libaums/blob/develop/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.java#L98 (should be the case when you are using my example app)

Now try to reproduce the problem. If it does not occur something is wrong with my FAT32 implementation. 
This fix above uses a different implementation for file systems (taken from jnode os, LGPG licensed). This is still a work in progress and should not be used in production.

magnusja, 2017-05-16:
In addition to that, I added a copy folder to USB option inside my example app. Can you try to reproduce that behavior there?

magnusja, 2017-05-16:
Can you try develop branch again for me, please?
I was able to reproduce and it might be that my last commit fixed the issue. Please check, thanks!

umair-brainx, 2017-05-16:
@alashow I have written a function that recursively copy all contents of a folder on internal storage to a folder on USB. That function works on some devices(Sony Xperia Z) correctly, on some devices(Samsung Note 4) it didn't copy any thing and on some devices(Samsung Tab 2) it copies a lot of junk files.

closed 2017-05-18 by magnusja