magnusja/libaums #80: How to directly read the files on the USB flash drive ?How to get to the path under the USB file?

liminwu0626, 2017-06-06:
I encountered the following questions,
[image]

magnusja, 2017-06-06:
This seems to be an error from android itself. https://android.googlesource.com/platform/libcore/+/jb-mr2-release/luni/src/main/java/libcore/io/IoBridge.java#388

Are you even using my library?

liminwu0626, 2017-06-07:
Yes ,do you mean that I can not get the path directly to the USB drive?

magnusja, 2017-06-07:
I mean it does not look that your code is calling anything related to my library. all my package names include 'libaums', there is no single libaums in you stacktrace!

It seems you are using the standard Android API. If you show some code I can give you some more input.

liminwu0626, 2017-06-07:
Oh,sorry,did not understand what you mean...Really thank you four your time... 
You see the following is the function call stack.
[image]
[image]
[image]

magnusja, 2017-06-07:
Okay, so this does not work. The standard Java `File` API and libaums are not compatible. So a Java `File` object cannot find something on the USB drive using libaums with a file path or similar.

Instead, you should use the `UsbFileInputStream` like so:
```
workbook = new HSSFWorkbook(new UsbFileInputStream(usbFile));
// or buffered (may increase performance)
workbook = new HSSFWorkbook(UsbFileStreamFactory.createBufferedOutputStream(usbFile, currentFs))
// if you don't have the current fs this shpuld be fine as well
workbook = new HSSFWorkbook(new BufferedInputStream(new UsbFileInputStream(file)))

```

Next time, please read the documentation more carefully. It is explained how to use Input/OutputStreams.

closed 2017-06-07 by magnusja

liminwu0626, 2017-06-07:
Okay,Thank you so much...

magnusja, 2017-06-09:
You are welcome, I hope it works now!