magnusja/libaums #82: download file large than 2G

Feximin, 2017-06-12:
Hi, Magnusja, recently I use libaums to download file from my camera, downloading will be failed if the file is great than than 2G, and the exception like below, any response will be appreciated. Thanks!

 java.lang.ArrayIndexOutOfBoundsException: length=72192; index=-65536 
at  com.github.mjdev.libaums.fs.fat32.ClusterChain.read(ClusterChain.java:112) 
at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.java:159) 
at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.java:84)

magnusja, 2017-06-12:
So you are copying from the USB drive to the Android phone?

Feximin, 2017-06-13:
Absolutely, I don't know whether my operation is right or wrong, but I followed the demo carefully, can you give me some advice? Thanks!

magnusja, 2017-06-13:
I have indeed the same problem, I will have a look. Thanks!

magnusja, 2017-06-13:
Hey there,

thanks for the bug report. Should be fixed in v0.5.4, please reopen if you still have problems.

Feximin, 2017-06-14:
Hey magnusja, still can not work well when copying file large than 2G,  and the same exception was thrown,   thanks for your time!

magnusja, 2017-06-14:
Hmm, interesting, I will do some tests later today.

magnusja, 2017-07-03:
I am so sorry, I don't know what exactly happened, but fixed in v0.5.5

closed 2017-07-03 by magnusja