magnusja/libaums #85: Array Index Out Of Bounds Exception

anggerpr, 2017-06-14:
I'm using this lib for reading file. I just copy from your test file. But it give me an error if I'm not plug the USB.
```
06-14 15:53:43.203 25151-25151/id.co.softorb.app.product E/AndroidRuntime: FATAL EXCEPTION: main
                                                                           Process: id.co.softorb.app.product, PID: 25151
                                                                           java.lang.RuntimeException: Unable to start activity ComponentInfo{id.co.softorb.app.product/id.co.softorb.app.product.Authentication}: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
                                                                               at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2195)
                                                                               at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2245)
                                                                               at android.app.ActivityThread.access$800(ActivityThread.java:135)
                                                                               at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1196)
                                                                               at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                               at android.os.Looper.loop(Looper.java:136)
                                                                               at android.app.ActivityThread.main(ActivityThread.java:5017)
                                                                               at java.lang.reflect.Method.invokeNative(Native Method)
                                                                               at java.lang.reflect.Method.invoke(Method.java:515)
                                                                               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:779)
                                                                               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:595)
                                                                               at dalvik.system.NativeStart.main(Native Method)
                                                                            Caused by: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
                                                                               at id.co.softorb.app.product.Authentication.discoverDevice(Authentication.java:271)
                                                                               at id.co.softorb.app.product.Authentication.onCreate(Authentication.java:113)
                                                                               at android.app.Activity.performCreate(Activity.java:5231)
                                                                               at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)
                                                                               at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2159)
                                                                               at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2245) 
                                                                               at android.app.ActivityThread.access$800(ActivityThread.java:135) 
                                                                               at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1196) 
                                                                               at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                               at android.os.Looper.loop(Looper.java:136) 
                                                                               at android.app.ActivityThread.main(ActivityThread.java:5017) 
                                                                               at java.lang.reflect.Method.invokeNative(Native Method) 
                                                                               at java.lang.reflect.Method.invoke(Method.java:515) 
                                                                               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:779) 
                                                                               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:595) 
                                                                               at dalvik.system.NativeStart.main(Native Method) 
```

anggerpr, 2017-06-14:
it work fine... just add throws this exception.
```
private void discoverDevice() throws Exception{
        UsbManager usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(this);
        devicea = devices[0];
        UsbDevice usbDevice = getIntent().getParcelableExtra(UsbManager.EXTRA_DEVICE);
        if (usbDevice != null && usbManager.hasPermission(usbDevice)) {
            Log.d(TAG, "received usb device via intent");
            setupDevice();
        } else {
            PendingIntent permissionIntent = PendingIntent.getBroadcast(this, 0, new Intent(ACTION_USB_PERMISSION), 0);
            usbManager.requestPermission(devicea.getUsbDevice(), permissionIntent);
        }
    }
```

closed 2017-06-14 by anggerpr

magnusja, 2017-06-14:
Hey there, 

you should not handle that via an Exception, instead, check the array length.

anggerpr, 2017-06-15:
The array length is 0 and index is 0. I just try with your example. 
And what should I do? but thank, its awesome libs.

magnusja, 2017-06-15:
If the array length is 0 then you should not access the 0th index of the array, obviously. A length of 0 means no device attached, therefore don't try to access it.