magnusja/libaums #86: Investigate on SD card issues

magnusja, 2017-06-29:
@fly7632785 please let us continue here.

Prior discussion: https://github.com/magnusja/libaums/issues/70#issuecomment-311617027

magnusja, 2017-06-29:
@fly7632785 so you are showing screenshots with a device of capacity around 32gb and the diskutil list command shows 8gb of capacity. How is that?

I tested with my SD card reader and it did not work at all. It does not even read from the device, but I do not really know why. I emailed the developers of Nexus Media Importer, maybe they have some hints.

fly7632785, 2017-06-29:
I don't know why it is 8g. May the information is not whole. But the most important is the sd card of the USB equipment doesn't have a MBR partition. I will try to format it or re-partition it. I am not sure it will work. 
I will go to bed now, cause it is deep night.
Hope you have a new process and development, good night

magnusja, 2017-06-29:
So if there is no MBR then usually it should log 'not a valid mbr partition table' and it does not even try to read partitions from it.

fly7632785, 2017-06-30:
yes, so we need test. But there are two app which can read directly. I don't know what tech they use.

magnusja, 2017-06-30:
Which is the other one?
I tried Total Commander with the USB Plugin with my card reader and it does not work.

fly7632785, 2017-06-30:
If It's not like 'not a valid mar partition' you said. Its  get(offset) of every 16 byte is  just always 0, so always continue to the end.
We compare it to the other USB storage which works, figure it out that it seems it doesn't have a MBR.
If it definitely has a MBR. What cause the partitionType is always 0 and not any one of the PartitionType Arrays

fly7632785, 2017-06-30:
The other one I meant is a normal USB mass storage, not the USB equipment with a SD card.

fly7632785, 2017-06-30:
The other apps which can read  are here(https://play.google.com/store/apps/details?id=com.estrongs.android.pop&hl=ja   ;   https://play.google.com/store/apps/details?id=com.homeysoft.nexususb.importer&hl=ja)

magnusja, 2017-06-30:
0 usually means empty partition table, see here: https://en.wikipedia.org/wiki/Partition_type

Are you sure that it is always 0? Maybe it is an extended partition, that is not supported at the moment as well.

fly7632785, 2017-06-30:
Yes, what I said is wrong. What I really mean is there is no partition instead of no MBR. 
Yes we debug and it is always 0 four times

magnusja, 2017-06-30:
Can you create an image for me like here: https://www.cyberciti.biz/faq/how-to-create-disk-image-on-mac-os-x-with-dd-command/

Then please upload and sent it to me. Then I can anaylze it. You might want to limit the size to a couple of 100 of MBs, I do not need the whole 8 or 32gb image.

fly7632785, 2017-06-30:
By the way, you are in Germany. It is about 3 am? Don't you sleep?

magnusja, 2017-06-30:
You can also use the disk utility app of OSX no need to use the command line.

I am currently in the US, it is 6pm here :P

fly7632785, 2017-06-30:
http://jafir-my-love.oss-cn-shanghai.aliyuncs.com/usb.img        
Is this ?
I create it by dd 
If it's useless.
I will try the method of the web

magnusja, 2017-06-30:
It is only 10 bytes, so literally nothing in there.

fly7632785, 2017-06-30:
OK I'll try, please hold on.
I am on the way to my company

fly7632785, 2017-06-30:
http://jafir-my-love.oss-cn-shanghai.aliyuncs.com/backup.my.sdcard.img.dd        about 98M

magnusja, 2017-06-30:
thanks, I will have a look asap!

fly7632785, 2017-06-30:
ok

fly7632785, 2017-06-30:
maybe the USB equipment is GPT not MBR
The 8 GB card is a test card(never mind)
[image]

magnusja, 2017-06-30:
Yes it seems like it is. Try reformatting with MBR.

fly7632785, 2017-06-30:
yes, I am trying to do that

fly7632785, 2017-06-30:
May I have your email ?

magnusja, 2017-06-30:
Sure, it is mjahnen <at> stanford.edu

fly7632785, 2017-06-30:
mjahnen@stanford.edu  ?

magnusja, 2017-06-30:
Yes, I just wrote it like that so no one can scrape the email and send me spam.

fly7632785, 2017-06-30:
ok    roger

fly7632785, 2017-06-30:
Hi, magnusja. 
if it's GPT, should the framework have a log notice?

magnusja, 2017-06-30:
According to Wikipedia there is a protective MBR for devices without GPT support. (https://en.wikipedia.org/wiki/GUID_Partition_Table#Protective_MBR_.28LBA_0.29)

That is why libaums recognizes an MBR. I analyzed the file and in fact, there are the magic numbers for an MBR the rest of the file are mostly zeros. So please reformat with MBR and FAT32 then you are good to go.

fly7632785, 2017-06-30:
I'm sorry to tell you the message.
When I convert it to MBR from GPT.
It still doesn't work....

fly7632785, 2017-06-30:
the result of debug is still same.

fly7632785, 2017-06-30:
[image]
this is screenshot

`bin-no-MBP:Volumes bingao$  hexdump -C -s 446 -n 64 NO\ NAME/
000001be  ff ff 54 52 41 53 48 45  7e 31 20 20 20 12 00 9b  |..TRASHE~1   ...|
000001ce  af 55 da 4a da 4a 00 00  af 55 da 4a bd 00 00 00  |.U.J.J...U.J....|
000001de  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001fe
`
this is MBR code

But It seems it is wrong.
The fifth byte each 16 is not 0.  But the result of debug is always 0.
Why is that?

magnusja, 2017-06-30:
This looks like you are already into the file system.
Are you sure you formatted with MBR, maybe the file system starts now directly?

fly7632785, 2017-06-30:
This is  a info from Hard Disk Manager about the USB equipment.
It shows it is MBR and FAT32.
I didn't format it (what I formatted was the test equipment before), it's just MBR
Is the command code right ?

magnusja, 2017-06-30:
You need to hexdump on the low-level device like /dev/diskX. I don't know what NO\ NAME is in your hexdump call, but it seems to be the directory of the device?

fly7632785, 2017-06-30:
yes, your command code is right

fly7632785, 2017-06-30:
`
bin-no-MBP:Volumes bingao$ diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *251.0 GB   disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:          Apple_CoreStorage Macintosh HD            250.1 GB   disk0s2
   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1 (internal, virtual):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                  Apple_HFS Macintosh HD           +249.8 GB   disk1
                                 Logical Volume on disk0s2
                                 8ABB2546-043F-4E17-A20A-8EECAB36BE00
                                 Unlocked Encrypted

/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:                            NO NAME                *31.9 GB    disk2

`

`
bin-no-MBP:Volumes bingao$ sudo hexdump -C -s 446 -n 64 /dev/disk2
000001be  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001fe
`

magnusja, 2017-06-30:
try hexdump -C  -n 512 and please make sure the formatting does not get lost.

fly7632785, 2017-06-30:
bin-no-MBP:Volumes bingao$ sudo hexdump -C -n 512 /dev/disk2
Password:
00000000  eb 00 90 20 20 20 20 20  20 20 20 00 02 40 92 04  |...        ..@..|
00000010  02 00 00 00 00 f8 00 00  3f 00 ff 00 00 20 00 00  |........?.... ..|
00000020  00 04 b7 03 b7 1d 00 00  00 00 00 00 02 00 00 00  |................|
00000030  01 00 06 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000040  80 00 29 31 63 31 62 4e  4f 20 4e 41 4d 45 20 20  |..)1c1bNO NAME  |
00000050  20 20 46 41 54 33 32 20  20 20 00 00 00 00 00 00  |  FAT32   ......|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 55 aa  |..............U.|
00000200

magnusja, 2017-06-30:
I wonder why it says FAT32, if I dump an MBR from a USB flash drive it is all 0s besides the last two bytes and the actual entries.

Can you try this:  `sudo dd if=/dev/disk2 of=/tmp/mbrsda.bak bs=512 count=1` and send the me file

fly7632785, 2017-06-30:
http://jafir-my-love.oss-cn-shanghai.aliyuncs.com/mbrsda.bak

fly7632785, 2017-06-30:
NO NAME maybe its name.

magnusja, 2017-06-30:
Okay, it is exactly the same and the partition type information is all zeros.
So there is something wrong. Either you still have an MBR or there is no boot sector at all??

Can you send some screenshots from the built in macOS disk utility tool, please?

fly7632785, 2017-07-01:
maybe  the sd card of the usb equipment doesn't have a mbr sector   known by  some books.

fly7632785, 2017-07-01:
for save space of storage

fly7632785, 2017-07-01:
[image]

https://stackoverflow.com/questions/38004064/is-it-possible-that-small-sd-cards-are-formatted-without-an-mbr

[image]

magnusja, 2017-07-01:
Could certainly be. So please reformat with MBR then.

fly7632785, 2017-07-01:
So, do you know any way to format the scared with MBR?

magnusja, 2017-07-01:
[image]

fly7632785, 2017-07-02:
[image]
it's not  Generic STORAGE DEVICE Media which has a function to choose a partition and format, instead of a NO NAME which just has choose Format, no Scheme.

what does that mean ?

magnusja, 2017-07-02:
Hmm I don't know. What does the thing above the "NO NAME" mean?

fly7632785, 2017-07-03:
Thank you, it works now.
I have sent the e-mail to you.

fly7632785, 2017-07-03:
[image]
erase it and reformat it to support MBR and FAT32,
it's OK

magnusja, 2017-07-03:
Nice, glad you could figure it out!
If you need something else, don't hesitate to ask!

closed 2017-07-03 by magnusja

fly7632785, 2017-07-26:
Hi, @magnusja 
I come again.
What I want to ask is that if there is any way to read data from SDCard without MBR.
Many equipments of the customer are not supported by MBR.

magnusja, 2017-07-27:
Hey @fly7632785 

I am not sure when I will be able to look into this. The biggest problem is, how can we distinguish between MBR and not MBR/FAT32, if FAT32 looks like MBR? 
If you would like to work on this, I'd be happy to take an PR!

fly7632785, 2017-07-28:
Yes, What I want to do is investigating whether we can format the SDcard by Android System.
But maybe it need Root or System permission.

magnusja, 2017-07-28:
Probably not possible without user interaction. In addition, to do that over the Android system, the user needs to plug the SD card into the phone directly (if the device does not support mounting USB drives natively). 
In theory, you could add this functionality to libaums, but then you could also just add support for drives without MBR.

magnusja, 2017-07-30:
Have a look at this branch: https://github.com/magnusja/libaums/tree/feature/device-without-mbr

I implemented acess for devices without mbr.

fly7632785, 2017-07-31:
Yes, that's amazing.
I am searching a way to access without MBR
Cause, there's another app which can read without MBR
So In theory, it's practicable

[11 more replies]