magnusja/libaums #90: UsbFile to File

SvenSv, 2017-07-11:
Hi,

how can I converst from UsbFile to java.io.File and vice versa?

nagesh-kashyap-r, 2017-07-11:
Hi,

Please help in converting UsbFile to File.
Or at least get path of the file on USB Drive.

magnusja, 2017-07-11:
Hey there,

this is not possible. What you can try is to create your own class and mimic java.io.File behavior using an UsbFile, although I would not recommend that.

You can also not get a file path to the USB drive because libaums cannot be integrated into the Android/Linux file (system) handling.

If you need to make files available to other applications or APIs which do not support InputStreams, you can try the http or storageprovider module explained on the main page. 

Other than that you might want to explain your problem in more detail so that I can further assist you.

Thanks,
Magnus

closed 2017-07-11 by magnusja