magnusja/libaums #95: java.io.IOException: Could not write to device, result == -1 errno 88 Socket operation on non-socket

SvenSv, 2017-07-26:
I get this error message, sometimes when I try to copy files from android to UsbDevice:

https://stackoverflow.com/questions/45329444/android-libaums-java-io-ioexception-could-not-write-to-device

Thanks for your help

magnusja, 2017-07-27:
Hey @SvenSv,

this is a known issue, but I haven't figured out yet what is wrong or how to investigate. For me, this error mostly occurs when multiple devices are connected.
I talked to the developer of USB Media Explorer and he is aware of this problem, but unable to solve it as well.  

> Yes, I'm painfully aware of #1.  According to the Mass Storage spec, you should be able to a Bulk-Only Mass Storage Reset and recover.  Unfortunately, it doesn't work or at least I could never get it work.  What I learned later on is that there is the Mass Storage spec and what Windows does.  A great number of (cheap) devices only work reliably if you do what Windows does.  FYI:  Windows does a port reset to recover from this.  I couldn't figure out how to do that without root.

magnusja, 2020-01-14:
see #175

closed 2020-01-14 by magnusja