magnusja/libaums #97: Provider -XML - unresolved package 'storageprovider'

SvenSv, 2017-07-31:
After I added the XML-Tag from the readme / how to use -> Storage Access Framework section:

<provider
            android:name="com.github.mjdev.libaums.storageprovider.UsbDocumentProvider"
            android:authorities="com.github.mjdev.libaums.storageprovider.documents"
            android:exported="true"
            android:grantUriPermissions="true"
            android:permission="android.permission.MANAGE_DOCUMENTS"
            android:enabled="true"> 
            <intent-filter>
                <action android:name="android.content.action.DOCUMENTS_PROVIDER" />
            </intent-filter>
</provider>

Andoid Studio complains about the first line: unresolved package 'storageprovider'.

What am I doing wrong? Thanks in advance.

magnusja, 2017-07-31:
Did you install the storageprovider module?

SvenSv, 2017-08-01:
I guess not, all I did was
compile 'com.github.mjdev:libaums:0.5.5'

How do I install the module?

SvenSv, 2017-08-01:
...
compile 'com.github.mjdev:libaums-storageprovider:0.5.1'     ???

closed 2017-08-01 by SvenSv