magnusja/libaums #98: UsbFile Metadata

SvenSv, 2017-08-01:
I have an UsbFile, which is a .mp3-file. How can I get the metadata for this UsbFile?

Thanks in advance!

magnusja, 2017-08-01:
You mean like artist and song title and all of that?
I am sorry, but that is way beyond the scope of this library.
I would suggest using another library for that, maybe you can try this one: https://github.com/beckchr/ithaka-audioinfo

closed 2017-08-01 by magnusja

SvenSv, 2017-08-02:
But these libraries want to check the metadata of an file object, not of an UsbFile, don't they? How do I get about using such libraries when I use your library and have the audio files as UsbFiles?

magnusja, 2017-08-02:
Did you even check the first example?
```
File m4aFile = new File("sample.m4a");

try (InputStream input = new BufferedInputStream(new FileInputStream(m4aFile))) {
	AudioInfo audioInfo = new M4AInfo(input);
	...
} catch (Exception e) {
	...
}
```

It is using an InputStream, which works with UsbFileInputStream. Some librarys only support java.io.File, which would not work with UsbFile, so you have to either use a library which supports InputStream or maybe HTTP like this one: https://github.com/wseemann/FFmpegMediaMetadataRetriever (But then you would need to use the HTTP module as well, so I would stick to InputStream)