magnusja/libaums #99: storageprovider: "content could not be loaded right now"

SvenSv, 2017-08-02:
I'm using the storageprovider to read some mp3 files, and when I select the usb-drive, all folders and files are shown. As soon as I select the "music" folder, an hourglass appears (as a background image, not a busy-aniamtion) with the text "content could not be loaded right now" ("Inhalte können momentan nciht geladen werden." in my german version).

Any idea whats happening?

SvenSv, 2017-08-02:
Also, when theres an mp3 file in the root folder, it is displayed, but not selectable (or at least, nothing happens on press) , even though mp3 files from internal storage are selectable.

magnusja, 2017-08-02:
Can you post a minimal example, which reproduces this error?

magnusja, 2017-08-03:
Are you using the stroageprovider and the normal API to the UsbMassStorageDevice at the same time? There are problems if two different processes are accessing the device simultaneously. Maybe try the storageprovider isolated and see if it works.

SvenSv, 2017-08-08:
sorry I didn't write earlier, I couldn't solve the problem and went another way.

magnusja, 2017-08-08:
Can you maybe explain what exactly you did, so that other people know how to solve this problem?

SvenSv, 2017-08-15:
Well, I'm afraid I just didn't show the files in my app. Now I have to show them and I'm back at the same problem. By now I have
    compile 'com.github.mjdev:libaums:0.5.5'
    compile 'com.github.mjdev:libaums-storageprovider:0.5.1'
    compile 'com.github.mjdev:libaums-httpserver:0.5.2'
in my gradle file and I jsut call the SAF with the usual

                        intent = new Intent(Intent.ACTION_OPEN_DOCUMENT);
                        int OPEN_REQUEST_CODE = 41;
                        intent.setType("*/*");
                        startActivityForResult(intent, OPEN_REQUEST_CODE);

magnusja, 2017-08-17:
Hey there,

so using the storageprovider and the normal library/http module at the same time eads to problems. This is because the storageprovider and the normal library both access the USB device from different processes or threads. I do not know why this leads to problems, unfortunately, but try to only use one at the same time.

Why do you think you need the http and the storageprovider module at the same time?

SvenSv, 2017-08-17:
It would have been nice for convenience. I stopped using the storageprovider.

magnusja, 2017-08-17:
So does that mean it works now better?