magnusja/libaums #102: efficiency problem.

SvenSv, 2017-08-08:
I use a FileServer to stream mp3s to a media player. When I hit the "next"-button, the next song plays, but it takes a while to start. The duration of the pause between two songs depends on the length of the first song.
It looks like the fileserver first streams the complete first mp3 and then opens the new song. 

onCreate:
            HttpServer server = new AsyncHttpServer(8000);
            UsbFile dir = musicFiles[0].getParent();
            fileServer = new UsbFileHttpServer(dir, server);
            try {
                fileServer.start();
            } catch (IOException e) { }

startPlay:
           try {
                player.setDataSource(this, Uri.parse(fileServer.getBaseUrl() + file.getName()));
                player.prepare();
                player.start();
            } catch (IllegalArgumentException | IllegalStateException | IOException e) { }

magnusja, 2017-08-08:
Hey @SvenSv,

I had some other people reporting long buffer times, but only with HD videos. I am not really sure what is going on there, unfortunately. You can maybe try to use NanoHTTP instead of AsyncHttp.

SvenSv, 2017-08-09:
Hey thanks! Switching from Async to NanoHTTP did the trick! 
thank you so much!

closed 2017-08-09 by SvenSv