magnusja/libaums #103: partitionType is 110    and    FAT32

XIONG-WEN, 2017-08-10:
[image]

magnusja, 2017-08-10:
Which version are you using and what partition type should it be?

magnusja, 2017-08-10:
Oh sorry, just saw that it should be FAT32. Are you using develop branch or 0.5.5?

XIONG-WEN, 2017-08-11:
Thank you for reply.
I am using 0.5.5.
I set type = 2 when partitionType == 116 .
[image]
[image]

I will get usb files.

XIONG-WEN, 2017-08-11:
@magnusja 
Hi, I have a problem to need your help.
I have an fat32 USB drive, but the partitionType is 116. In your codes can not mapping out 'fat32', so it doesn't work.

magnusja, 2017-08-12:
Hey there,

so 110 in hex is 0x6E and 116 in hex is 0x74. If you have a look on Wikipedia (https://en.wikipedia.org/wiki/Partition_type) and look up these values you can see that they are not usually referring to FAT32.

I would suggest to reformat it and see if it works after that.
What you also can try is to use the current develop branch, I fixed a bug while reading the partition type, but I doubt that this bug is affecting you. 

Where did you get this drive? Is there something special about it? 

Another source: https://www.win.tue.nl/~aeb/partitions/partition_types-1.html

XIONG-WEN, 2017-08-13:
Thank you!
The new version is no problem.

closed 2017-08-13 by XIONG-WEN