magnusja/libaums #108: Feature Request: Expose ByteBlockDevice to public API

mrolcsi, 2017-08-18:
I've been using your library in [my own](https://github.com/mrolcsi/FileBrowser-Android) for a while now, and it works great!

But in a different, work related project, I need to read from and write to the USB device *directly*, bypassing the file system and partition table. So far I managed to do this by exposing the `BlockDeviceDriver` member from `UsbMassStorageDevice` with a getter, then using `blockDevice.read(offset, buffer)` and `blockDevice.write(offset, buffer)` to access the device. This works fine, but I need to use my own fork to make this work, and this makes it harder for me to update the fork with your changes.

I see that you made your own implementation to achieve the same thing (ByteBlockDevice), and so I ask you: can you make this available to the API?

If you want I can make a PR with the changes, I just thought it's better to ask first.

Thanks in advance!

magnusja, 2017-08-19:
Hey there,

why would you need access to that on such a low level?
I am not sure if that is a good idea because writing to the raw block device may result in the file system and the actual content on the drive being out of sync. 
If you really need that, just use the getUsbDevice and then use `UsbCommunicationFactory` to create a block device.
The ByteBlockDevice is just a wrapper to convert from byte offset to device offset. (file systems usually use byte offset addressing)

mrolcsi, 2017-08-29:
*(Sorry for the late answer, I was on vacation.)*

You're right, it's a pretty special case. To be clear, I don't want to use the filesystem at all. I'm just storing some data on the device as is. That's because I don't want the end user to know how the data I'm storing is structured. (It's also encrypted and stuff, but that's a whole other thing.)

I tried using `UsbCommunicationFactory` as you suggested, but I still need to determine the right UsbInterface to use (if there's more than one), then init a block device with `BlockDeviceDriverFactory.createBlockDevice()` before being able to use it. All of this already happens in `UsbMassStorageDevice`. I could just copy that part of your code to mine as needed, but I don't think that's the right approach.

All I need basically is a getter for the `blockDevice` inside `UsbMassStorageDevice`, and not to fail initialization if there are no partitions. (Right now it fails with a `NullPointerException` when trying to access a null `partitionTable`.) Then I could just use the already initialized `blockDevice` without any extra code. This would only need a small amount of changes in both your code, and mine.

magnusja, 2017-09-08:
Hey there,

so I am still not a 100% convinced to make this public API because it is an edge case. I rather like to keep that kind of stuff private.
The problem with the `NullPointerException` is certainly a problem. If `partitionTable == null` then it should not crash but rather just init with zero partitions. I would be happy to merge a PR for that.
If the `init()` call does not crash anymore, you can maybe use java reflections to get the `blockDevice`? Would that be an option?

mrolcsi, 2017-09-12:
Huh, I haven't thought about reflection. Yeah, that could work.
Right now I'm a little busy with other stuff, but I'll do a PR as soon as I can. Thanks!

mrolcsi, 2017-09-26:
I finally got around to check this `NullPointerException` and it seems that it's only present on the master branch. On develop `init()` throws an `IOException` ([here](https://github.com/magnusja/libaums/blob/33f67b6a65be6b3677abd688d3ee57632764addb/libaums/src/main/java/com/github/mjdev/libaums/partition/PartitionTableFactory.java#L76)) which gets caught in MainActivity ([here](https://github.com/magnusja/libaums/blob/33f67b6a65be6b3677abd688d3ee57632764addb/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.java#L776)), so the app does not crash.

Seems to me you already fixed this in #96

mrolcsi, 2017-10-03:
Update: Here's a PR that fixes the NullPointerException in the master branch: #113

magnusja, 2019-08-26:
This is implemented now :)

closed 2019-08-26 by magnusja