magnusja/libaums #110: UsbMassStorageDevice init() sometimes delays for minutes

cumbie, 2017-09-14:
After requesting the user for permission to the USB drive, I enumerate devices using UsbMassStorageDevice.GetMassStorageDevices(context).  I then get the device with the matching DeviceId of the one that requested permission.  I then call the init() method for the device.  Usually it executes immediately and I have access to the drive with no problems.  However, sometimes it will hang at the init() call for at least 2 minutes or more, and then work correctly.  This usually happens rarely, but I have found that I am able to reproduce this more often if I wait about 20+ seconds to click Ok on the permission dialog.

Any ideas what may be causing this?  I am testing with a Samsung Galaxy J7 phone, running Android OS 7.0.

magnusja, 2017-09-15:
Hey there,
no, I have not experienced that so far. Do you have the same problem for other devices as well?

cumbie, 2017-09-16:
The only other device I've tested with was a Samsung Galaxy S5 running Android 5.0, and I did not see this problem.  Not really enough testing to conclude yet, but it could be a problem just with OS 7.0.  Are you able to test and try to reproduce this problem?

magnusja, 2017-09-18:
I only have a SOny Xperia Z2 Tablet with Android 6. 

Newer Samsung phones/Android versions seem to make problems with this library. There is also the problem that sometimes created files are not actually written to the drive. Unfortunately, I haven't had time so far to look into this and buy a Samsung device.